arduino/ArduinoCore-avr #7: The repository has no tags

per1234, 2018-02-27:
Please add tags for each Arduino AVR Boards release. I can provide the necessary `git tag` commands if that would be helpful.

matthijskooijman, 2018-02-28:
Someone in #arduino just asked for a changelog, and I had a hard time figuring out if this repository was actually being used because I could find no tags or releases.

In addition to having tags, it would also help to have release descriptions with changelogs, like https://github.com/arduino/Arduino/releases used to have for the AVR cores.

On a related note: 1.6.21 is available in the board manager, but I can't find a commit for that here. What does that mean? Is this repo even used for releases?

On another related note: This repo seems to have a bunch of (unused?) branches, perhaps those should be cleaned up?

facchinm, 2018-03-02:
We really need to clean things up and complete the transition of PRs (not related to chainsaw) to this repo. The "*master" branches can be wiped but we are severely outnumbered to do all this stuff. If anyone is interested in 
* double checking the main Arduino repo for missing PRs related to AVR core
* reopen the PRs here using the "pr_*" branches, referencing the old ones
* closing the old ones
we could finally merge https://github.com/arduino/Arduino/commit/339467eec439576408e10f934ba839bd119fe56a into master and clean up the whole situation.

matthijskooijman, 2018-03-02:
Can't we merge that commit before all pending PR's are transitioned? Nothing changes about those PR's, except that they cannot be directly merged without additional work (but that is also the case now, when merging AVR pr's they must be duplicated here).

facchinm, 2018-03-02:
@cmaglie what do you think about that?

NicoHood, 2018-09-02:
Any updates on this issue?

wmacevoy, 2019-01-07:
I need to find the differences between 1.6.21 and 1.6.22 - something changed that affects the arduinounit library.  Tags would really help here....

facchinm, 2019-01-07:
@wmacevoy here we are https://github.com/arduino/ArduinoCore-avr/compare/1.6.21...1.6.22 (I double checked with the actual packages on the download server to make sure the content is the same).
1.6.21 was the last version created from the IDE repo, and also the last one to require `gcc-4.9.2-atmel3.5.4-arduino2` (I think this last information might be the one you need most)

lxrobotics, 2019-09-18:
@per1234 As far as I can see there are now tags provided for each release in this repository. Does this resolve your issue? If so please close it.

matthijskooijman, 2019-09-19:
I see now four tags, which looks good. However, I wonder if we also need tags for older releases (that were made when this code was still in the Arduino/Arduino repo), since AFAICS we do have the history for those? Might be non-trivial to figure out what revisions to tag, though.