arduino/ArduinoCore-avr #9: Feature Request: Add Support for AVR Dragon

sktrea, 2018-03-02:
Hi Could you add support for the Atmel (Microchip) AVR Dragon programmer?
(moved from Arduino Issues #7269)

MCUdude, 2018-03-09:
I have an AVR dragon I tried to use with Arduino IDE a few years ago.

When you hit _Burn Bootloader_ you're actually issuing two avrdude commands. the first one sets the fuses, and the second one loads the bootloader hex file. The problem I faced was that the programmer automatically reset itself after the fuses were sat. When the next command were executed immediately after, my computer couldn't find the USB device. If a small delay were added between the commands, it would work just fine.

facchinm, 2018-03-12:
I remember we already saw and fixed the "second command sent too fast" issue maybe a couple of years ago. I can't find the commit, though, so maybe it's worth checking again if the fix is actually in place.

per1234, 2018-03-12:
I think this is the relevant issue report: https://github.com/arduino/Arduino/issues/2986. That issue was primarily about the AVRISP mkII but there is also discussion of what appears to be the same problem for the Dragon in the thread.

https://github.com/arduino/Arduino/issues/2986 was solved for me by the update to avrdude 6.3.0. I also was able to separately solve it by changing from the libusb-win32 to libusbK driver (even when using avrdude versions prior to 6.3.0). However, there is a [report ](https://github.com/arduino/Arduino/issues/2986#issuecomment-248947514)of the problem still occurring on for the Dragon that issue thread.

MCUdude, 2018-03-21:
I just succeeded rising my Dragon up from the dead! The power circuity was faulty (extremely poor design), but some reflowing did the trick!

Here's the output when trying to burn bootloader to an ATmega32 using the Dragon in ISP mode:

```
/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino12/bin/avrdude -C/Users/Hans/Documents/Arduino/hardware/MightyCore/avr/avrdude.conf -v -patmega32 -cdragon_isp -Pusb -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0xd6:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/Hans/Documents/Arduino/hardware/MightyCore/avr/avrdude.conf"
         User configuration file is "/Users/Hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : dragon_isp
avrdude: usbdev_open(): Found AVRDRAGON, serno: 00A200000F1A
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              1
S_MCU:
  boot-loader FW version:        255
  firmware version:              7.39
  hardware version:              2
Serial number:                   00:a2:00:00:0f:1a
Device ID:                       AVRDRAGON
         AVR Part                      : ATmega32
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    10    64    0 no       1024    4      0  9000  9000 0xff 0xff
           flash         33     6    64    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00

         Programmer Type : DRAGON_ISP
         Description     : Atmel AVR Dragon in ISP mode
         Vtarget         : 5.0 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude: Device signature = 0x1e9502 (probably m32)
avrdude: erasing chip
[... 62 lines cut ...]

/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino12/bin/avrdude -C/Users/Hans/Documents/Arduino/hardware/MightyCore/avr/avrdude.conf -v -patmega32 -cdragon_isp -Pusb -Uflash:w:/Users/Hans/Documents/Arduino/hardware/MightyCore/avr/bootloaders/optiboot_flash/atmega32/optiboot_flash_atmega32_UART0_115200_16000000L.hex:i -Ulock:w:0x0f:m 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/Hans/Documents/Arduino/hardware/MightyCore/avr/avrdude.conf"
         User configuration file is "/Users/Hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : dragon_isp
avrdude: usb_open(): cannot read serial number "Input/output error"
avrdude: usb_open(): cannot read product name "Device not configured"
avrdude: usbdev_open(): Found [unnamed product], serno: [unknown]
avrdude: usbdev_open(): error claiming interface 0: Device not configured
avrdude: usbdev_open(): no usable interface found
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2107)

avrdude done.  Thank you.

Error while burning bootloader.

```

mcuee, 2022-06-04:
This will be a nice feature to add.

nmaas87, 2022-06-04:
@per1234 made an awesome POC some years ago and got it working: https://github.com/per1234/InoAVRDragon

mcuee, 2022-06-04:
Unfortunately it does not seem to work in the 1.8.19 version of Arduino IDE.