arduino/ArduinoCore-avr #10: Nano with Optiboot: maximum sketch size and fuses not updated

oqibidipo, 2018-03-08:
If I am not mistaken, these
```
nano.menu.cpu.atmega328.upload.maximum_size=30720
nano.menu.cpu.atmega328.bootloader.high_fuses=0xDA
```
should be
```
nano.menu.cpu.atmega328.upload.maximum_size=32256
nano.menu.cpu.atmega328.bootloader.high_fuses=0xFD
```

facchinm, 2018-03-09:
Hi @oqibidipo , you are right, but the boards in production have been flashed with the old fuse setting, making the bootloader partition larger than needed (see https://github.com/arduino/Arduino/commit/1cf34c8250658c5c4a5c488f7a7a0d7071e3430f#commitcomment-27651798). So we can't change the maximum sketch size parameter without affecting the functionality on these boards, sorry.

closed 2018-03-09 by facchinm

pfeerick, 2020-06-29:
Wouldn't adding another entry for the Nano been a better solution for this? If the production units are still being incorrectly flashed, call it (New Bootloader - Production) and (New Bootloader - Max Flash) or something like that.