arduino/ArduinoCore-avr #11: Rebuilding everything

iglosiggio, 2018-04-10:
Hi, i'm currently [making a package](https://github.com/HuayraLinux/pkg-arduino-avr-core) for a debian based distribution (and hoping to make that package go to debian itself) and when possible packaging binaries is a no-no.

Under the bootloaders folder there are a lot of `.hex` files that i don't know how to rebuild (missing USB_ID variable on the makefile).

Is documented anywhere how to rebuild all the core? I've downloaded the old versions of lufa and rebuilt most of the things that are here.

Related question. Is everything in the core needed? For example the bootloader `Leonardo-prod-firmware-2012-04-26.hex`

iglosiggio, 2018-04-10:
Related question, how can i rebuild the wifishield firmwares w/o a visual studio license?

iglosiggio, 2018-04-11:
If i upload builds with recent gcc versions do you have the necessary hardware to test the firmwares and bootloaders?

facchinm, 2018-04-12:
We have the hardware but the time available for these kind of tasks is very limited; normally the rebuilder should take care of testing the builds, or you could release a beta version of the deb and let the community test it and report if the firmwares are ok.

iglosiggio, 2018-04-12:
On my side i have the time but not the hardware, it seems like community testing is the way to go.

iglosiggio, 2018-04-12:
Update! :smiley:
I've managed to build the following bootloaders:

* bootloaders/bt/ATmegaBOOT_168_atmega328_bt.hex
* bootloaders/atmega8/ATmegaBOOT.hex
* bootloaders/lilypad/LilyPadBOOT_168.hex
* bootloaders/caterina-LilyPadUSB/Caterina-LilyPadUSB.hex
* bootloaders/atmega/ATmegaBOOT_168_atmega328_notp.hex
* bootloaders/atmega/ATmegaBOOT_168_ng.hex
* bootloaders/atmega/ATmegaBOOT_168_lilypad.hex
* bootloaders/atmega/ATmegaBOOT_168_atmega328.hex
* bootloaders/atmega/ATmegaBOOT_168_pro_20mhz.hex
* bootloaders/atmega/ATmegaBOOT_168_pro_16MHz.hex
* bootloaders/atmega/ATmegaBOOT_168_lilypad_resonator.hex
* bootloaders/atmega/ATmegaBOOT_168_pro_8MHz.hex
* bootloaders/atmega/ATmegaBOOT_168_atmega328_pro_8MHz.hex
* bootloaders/atmega/ATmegaBOOT_168_atmega1280.hex
* bootloaders/atmega/ATmegaBOOT_168_diecimila.hex
* bootloaders/optiboot/optiboot_lilypad.hex
* bootloaders/optiboot/optiboot_pro_16MHz.hex
* bootloaders/optiboot/optiboot_atmega8.hex
* bootloaders/optiboot/optiboot_luminet.hex
* bootloaders/optiboot/optiboot_atmega644p.hex
* bootloaders/optiboot/optiboot_atmega168.hex
* bootloaders/optiboot/optiboot_atmega1280.hex
* bootloaders/optiboot/optiboot_pro_20mhz.hex
* bootloaders/optiboot/optiboot_lilypad_resonator.hex
* bootloaders/optiboot/optiboot_pro_8MHz.hex
* bootloaders/optiboot/optiboot_atmega328.hex
* bootloaders/optiboot/optiboot_atmega328_pro_8MHz.hex
* bootloaders/caterina-Arduino_Robot/Caterina-Robot-Control.hex
* bootloaders/caterina-Arduino_Robot/Caterina-Robot-Motor.hex
* bootloaders/stk500v2/stk500boot_v2_android2561.hex
* bootloaders/stk500v2/stk500boot_v2_penguino.hex
* bootloaders/stk500v2/stk500boot_v2_mega1280.hex
* bootloaders/stk500v2/stk500boot_v2_mega2560.hex
* bootloaders/stk500v2/stk500boot_v2_cerebotplus.hex
* bootloaders/stk500v2/stk500boot_v2_amber128.hex
* bootloaders/caterina/Caterina-Leonardo.hex
* bootloaders/caterina/Caterina-Esplora.hex
* bootloaders/caterina/Caterina-Micro.hex

A lot of these builds were done by `make <board>` and hoping that the makefile didn't mess up with the parameters.

I'm now wanting to make the COMBINED firmware (usbdfu+usbserial) and i will not focus on the wifishield firmware (i don't have time to port the project to a Makefile :disappointed:)

Can you give me any guidance on how the original 8U2 and 16U2 firmwares were built? How do you _combine_ usbdfu and usbserial?

Thanks a lot for your previous responses!

facchinm, 2018-04-13:
Great! The combined `hex` are nothing more than two binaries (in hex format) merged together (as simple as a `cat fw2.hex >> fw1.hex`).
So the procedure is:
- build the dfu project
- build the actual usb-serial fw
- cat them together

And you should obtain something similar to the provided combined hex

iglosiggio, 2018-04-13:
hahaha i thought that it was a more involved task