arduino/ArduinoCore-avr #15: A 16u2 board

DavidDGrande, 2018-05-15:
I saw the Descriptors.h on the Caterina.h file..Then I thought I could make a board of it but how am I going to adapt the Caterina bootloader to the 16u2. I would like to use multiplexers in circuit to increase IOs but It is quite a big problem having few of them working as both Input and Output...So my questions are "How can I adapt the Caterina to 16u2 and how can I increase IOs?

NicoHood, 2018-05-15:
https://github.com/NicoHood/HoodLoader2

facchinm, 2018-05-16:
As Nico pointed out, Hoodloader2 is a dual mode firmware that you can use as USB-to-serial converter and also program directly. 
Both the Caterina bootloader and Hoodloader are based on https://github.com/abcminiuser/lufa , which is the common ground for many projects using USB AVRs. If you want to start clean, pick LUFA and adapt it to your needs, otherwise Hoodloader is a very good choice :slightly_smiling_face:

closed 2018-05-16 by facchinm

DavidDGrande, 2018-05-22:
Oh Thank You for your immediate replies this might be good enough for my needs...