arduino/ArduinoCore-avr #20: String class ::remove has overlapping strncpy, undefined results per standard specs

earlephilhower, 2018-07-18:
The ESP8266 Arduino port just discovered the following line in `String::remove` where `strncpy` is used with overlapping source and destination addresses (it's moving things backwards in the allocated buffer).  See [link](https://github.com/esp8266/Arduino/pull/4932) for the valgrind warnings reported by our CI system.

https://github.com/arduino/ArduinoCore-avr/blob/b7c607663fecc232e598f2c0acf419ceb0b7078c/cores/arduino/WString.cpp#L697-L699

`str*cpy` is [undefined when you overlap source and destination](https://en.cppreference.com/w/c/string/byte/strncpy)

While it's true that on the AVR, an 8-bit MCU, str*cpy will be implemented in a byte-wise manner, this code has and will probably be copied to 16- and 32-bit systems where they may use optimizations which give strange results occasionally due to the overlap.  A simple replacement of the `strcpy` with a `memmove` in this line clears this error and provides desired behavior on all systems.

earlephilhower, 2018-07-18:
Similar problem with the `::trim` method:
https://github.com/arduino/ArduinoCore-avr/blob/b7c607663fecc232e598f2c0acf419ceb0b7078c/cores/arduino/WString.cpp#L722-L727

Replacing `memcpy` with `memmove` on line 727 will fix this undefined usage on all systems.

facchinm, 2018-07-19:
@earlephilhower wow, thanks for sharing, we'll investigate if it applies to other 32bit cores and apply the fixes. Thanks a lot!

facchinm, 2018-12-03:
Applied in https://github.com/arduino/ArduinoCore-API/commit/b99609eb6d0dd2b0c29ba1af506454fbaca1d0c2 , now it's a matter of moving cores to API asap :slightly_smiling_face:

closed 2018-12-03 by facchinm