arduino/ArduinoCore-avr #23: stkv500v2 bootloader corrupts data when reading EEPROM

mlasevich, 2018-07-27:
READ_EEPROM command treats each address as 16bit instead of 8bit, as a result, the output of read EEPROM is corrupt, giving every other byte.

mcuee, 2022-06-03:
This may be related. It does not seem to be an avrdude issue but rather the issue of the bootloader.
* https://github.com/avrdudes/avrdude/issues/515

mcuee, 2022-06-12:
I can confirm that #24 fixed this issue.

mcuee, 2023-06-09:
For those who want to use an Arduino Core with the fix, you can use this one.
https://github.com/MCUdude/MegaCore

It has incorported #24 fix here.
https://github.com/MCUdude/MegaCore/commit/eba4f4a827aa3597b8e65144caee83f69724f280

mcuee, 2023-06-10:
The zip file here contains this EEPROM fix and a few other fixes.
* https://github.com/avrdudes/avrdude/issues/1379#issuecomment-1585533005

per1234, 2023-06-11:
If I understand correctly, https://github.com/arduino/ArduinoCore-avr/issues/129 is a duplicate of this issue. Since https://github.com/arduino/ArduinoCore-avr/issues/129 has a bit more information, I'll close this one in favor of the other.

closed 2023-06-11 by per1234