arduino/ArduinoCore-avr #30: SoftwareSerial breaks abs macro

florian-schweiger, 2018-08-10:
When I include `SoftwareSerial.h`, other libraries break because the `abs(x)` macro stops working. One particular library that stops functioning properly is [uStepper](https://github.com/uStepper/uStepper).

The reason for it seems to be a section at the bottom of `SoftwareSerial.h` that undefines several macros:
```
// Arduino 0012 workaround
#undef int
#undef char
#undef long
#undef byte
#undef float
#undef abs
#undef round
```
I've commented this out in my project, and things are back to normal.

So I'm wondering, is this workaround still needed in the post-Arduino-0012 era?

matthijskooijman, 2018-11-24:
I dug around a bit to see if I could find the origin of these lines, but it seems they were already present in the ["NewSoftSerial" library](http://arduiniana.org/libraries/newsoftserial/) that was merged into the Arduino core a long time ago (and renamed to "SoftwareSerial" there. That library was again based on ladyada's AFSoftSerial, but that did not have these lines. Unfortunately, it seems NewSoftSerial did not use any version control, and the changelog does not mention these either.

In any case, I can't quite see why these undefs are needed. They are at the bottom of SoftwareSerial.h, so they cannot affect the contents of SoftwareSerial.h above. They might be needed for SoftwareSerial.cpp, but then they should be moved there, and not "pollute" the global scope. However, you suggest things work just fine with these line removed, so I'd be in favor of removing them (the more predictable the global scope is, the better).

@cmaglie, @facchinm, would you agree?

@florian-schweiger, care to prepare a PR for this?

facchinm, 2018-11-26:
@matthijskooijman agree, let's get rid of that lines :slightly_smiling_face:

PR #51 (2019-05-16): Remove historic Arduino 0012 workaround in SoftwareSerial
Remove `#undef`s in SoftwareSerial.h marked as _Arduino 0012 workaround_ that broke several macros, including `abs`.

Fixes #30.
closed 2019-05-16 by facchinm