arduino/ArduinoCore-avr #39: AVR core 1.6.22 (IDE 1.8.6) crashes avr-gcc compiler (5.4.0) when F() string is assigned to a static variable

bxparks, 2018-09-03:
The following program will crash the 5.4.0 avr-gcc compiler used in AVR core 1.6.22 (IDE 1.8.6). The problem does not happen in AVR core 1.6.21 (IDE 1.8.5):
```
void setup() {}

void loop() {
  static const __FlashStringHelper* F_STRING = F("fstring");
  Serial.println(F_STRING);
}
```

The error message is:
```
lto1: internal compiler error: in lto_output_varpool_node, at lto-cgraph.c:624
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper: fatal error: /home/brian/dev/arduino-1.8.6/hardware/tools/avr/bin/avr-gcc returned 1 exit status
compilation terminated.
/home/brian/dev/arduino-1.8.6/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Nano.
```

Interestingly, the following alternatives do **not** cause problems:
```
void setup() {}

static const char P_STRING[] PROGMEM = "pstring";

void loop() {
  Serial.println((const __FlashStringHelper*) P_STRING); // ok
  Serial.println(F("inline fstring")); // ok
}
```

My environment is Ubuntu Linux 17.10, but another person on the [Arduino Forum thread](https://forum.arduino.cc/index.php?topic=566280.0) reports the same error on Windows 10.

bxparks, 2018-09-03:
A workaround that works for me is to declare the `F_STRING` as a const, not a non-const function static, like this:
```
void loop() {
  const __FlashStringHelper* const F_STRING = F("fstring");
  Serial.println(F_STRING);
}
```

That is what I had actually intended in my real code, but I seem to have gotten confused by the "static const" part in front of a pointer type.

bxparks, 2018-09-04:
From the above mentioned Arduino forum thread, here's another example that produces the same compiler error, but without using a `static` keyword:
```C++
void loop() {
  const __FlashStringHelper* strings[] = { F("a"), F("b") };
  for (int i = 0; i < 2; i++) {
    Serial.println(strings[i]);
  }
}
```

My guess is that the compiler is performing an internal optimization that causes `strings` to become static. If the code is slightly rewritten, the compiler is happy:
```C++
void loop() {
  const __FlashStringHelper* const a = F("a"); 
  const __FlashStringHelper* const b = F("b");
  const __FlashStringHelper* strings[] = { a, b };
  for (int i = 0; i < 2; i++) {
    Serial.println(strings[i]);
  }
}
```

matthijskooijman, 2018-09-05:
Note that the `PSTR()` macro (and thus also `F()`) implicitly creates a static variable too.

I wonder what happens if we replace the `PSTR()` macro with the one shown here: https://github.com/mmurdoch/arduinounit/blob/5d16d578008f665f4ec0a9cc0818423cf45ee1c9/src/ArduinoUnitUtility/Flash.h#L11-L27

That version of `PSTR()` is intended to fix deduplication of flash strings (Arduino/arduino#4863), but it also uses a completely different way of generating the string variables, so I suspect it might not trigger this internal compiler error. Aynyone care to test? This would involve copying the linked code into, I think, `WString.h` and update the `F()` macro to use it.

Floessie, 2018-10-05:
Great findings and analysis, @bxparks.

I wonder, if anyone has upstreamed this ICE (and also [Arduino issue 7949](https://github.com/arduino/Arduino/issues/7949))... If so, I'd like to see a link to the upstream issue to follow it up there.

AFAIK, those problems were tackled in GCC 7 something, but the official Atmel toolchain lags behind (one can replace it with an up-to-date [crosstool-ng](https://github.com/crosstool-ng/crosstool-ng) generated, though).

Best,
Flössie

Floessie, 2018-11-22:
Ping. Any news or plans?

facchinm, 2018-11-30:
@Floessie can you take a look at https://github.com/arduino/Arduino/issues/7949#issuecomment-442969335 ? Thanks!

Floessie, 2018-11-30:
@facchinm Okay, I installed 1.8.7 and upgraded the AVR core to 1.6.208. I've not yet run into the ICE, so I test-compiled some of my projects to see if they are still working: All went fine with the 7.3 toolchain. :+1: 

As I had an ATTiny 841 at hand I also installed ATTinyCore and ran a build. No problems. :+1:

Also compiled with MiniCore and MightyCore for the 328 and 1284. Again, nessun problema. :+1: 

For parallel installation of 1.8.7 and 1.8.5 I moved the old `~/.arduino15` out of the way. To my surprise running 1.8.7 cleared my `~/Arduino` folder. Luckily, I had a backup. :-1:

All in all: Splendid. :smiley: Especially since the Tiny is still working (other than you expected).

HTH,
Flössie

MartyMacGyver, 2019-09-08:
I've been chasing this bug around this weekend as it cropped up in code I'm working on.

I see various conversations at https://github.com/arduino/Arduino/issues/7949 and https://github.com/arduino/Arduino/issues/8543 related to this (among others), all of which point to using the Arduino AVR 1.6.208 / 1.6.209 tools from http://downloads.arduino.cc/packages/package_avr_7.3.0_index.json as a workaround.

**When is this fix actually planned to be in a board package release?**

jfitter, 2019-09-19:
Arduino v1.8, Visual Micro, avrgcc 5.4.0-atmel3.6.1-arduino2
This problem caused me a lot of lost time until I made this change
`return pgm_read_byte_near(character_table + c);`
changed to 
`return pgm_read_byte(character_table + c);`
Not quite sure why but the problem has not returned.

facchinm, 2019-09-19:
I'm closing this issue since it's fixed by avr core 1.8.1 (with gcc 7.4.0)

closed 2019-09-19 by facchinm