arduino/ArduinoCore-avr #45: Old version in Boards Manager, Arduino as ISP not working

AndreKR, 2018-10-05:
I am using Arduino 1.8.5 and I have updated my Arduino AVR Boards core in the Boards Manager to the latest available version, which is 1.6.21.

With this setup, using an Arduino as ISP doesn't work and from what I found out I don't understand how it could ever have worked.

programmers.txt contains a section that sets:
```
arduinoasisp.protocol=stk500v1
```
However, the ArduinoISP sketch does not work with this setting, it has to be changed to
```
arduinoasisp.protocol=arduino
```

In fact I found a [commit](https://github.com/arduino/ArduinoCore-avr/commit/b084848f2eaf9ccb3ac9a64ac5492d91df4706bf) in master that makes that very change.

Despite this commit being 11 months old by now it doesn't seem to be pushed to installations via the Boards Manager.

per1234, 2018-10-05:
Outdated Arduino AVR Boards version in Boards Manager also reported here:
https://github.com/arduino/Arduino/issues/8062

We have the opposite complaint, saying the new `arduinoasisp.protocol=arduino` doesn't work, here:
https://github.com/arduino/Arduino/issues/8032

AndreKR, 2018-10-06:
Closing this as a duplicate for now. I might investigate about the meaning of `not in sync: resp=0x15` (https://github.com/arduino/Arduino/issues/8032) vs. `not in sync: resp=0x03` (my case).

closed 2018-10-06 by AndreKR