arduino/ArduinoCore-avr #49: HardwareSerial optimize circular buffer pointer overflow handling

hattesen, 2018-11-13:
The current [HardwareSerial.cpp](../blob/master/cores/arduino/HardwareSerial.cpp) uses a very inefficient modulo operator to wrap the head and tail pointers to the circular receive and transmit buffers in several places. This operation involves a slow division operation (around 200 clock cycles), which will severely limit the possible bandwidth (effective baud rate) of the serial communication.

A [blog post](https://mekonik.wordpress.com/2009/03/02/modified-arduino-library-serial/) points out that this (similar) change will not only more than double the effective baud rate (to 1Mbps), but also save 124 bytes of program memory.

```cpp
void HardwareSerial::_tx_udr_empty_irq(void)
...
  _tx_buffer_tail = (_tx_buffer_tail + 1) % SERIAL_TX_BUFFER_SIZE;
```
```cpp
int HardwareSerial::available(void)
{
  return ((unsigned int)(SERIAL_RX_BUFFER_SIZE + _rx_buffer_head - _rx_buffer_tail)) % SERIAL_RX_BUFFER_SIZE;
```
```cpp
int HardwareSerial::read(void)
  ...
    _rx_buffer_tail = (rx_buffer_index_t)(_rx_buffer_tail + 1) % SERIAL_RX_BUFFER_SIZE;
```
```cpp
size_t HardwareSerial::write(uint8_t c)
  ...
  tx_buffer_index_t i = (_tx_buffer_head + 1) % SERIAL_TX_BUFFER_SIZE;
```
... one pointer wrap/overflow handling is done conditionally (but could still be simplified):

```cpp
int HardwareSerial::availableForWrite(void)
  ...
  if (head >= tail) return SERIAL_TX_BUFFER_SIZE - 1 - head + tail;
    return tail - head - 1;
```

The buffer pointers should be wrapped using fast bitwise-and operation or conditional addition/reset. Since the buffer size cannot be guaranteed to always match a power-of-two, bitwise-and are not usable, so I suggest using

```cpp
void HardwareSerial::_tx_udr_empty_irq(void)
...
  _tx_buffer_tail = _tx_buffer_tail < SERIAL_TX_BUFFER_SIZE - 1  ? _tx_buffer_tail + 1 : 0;
```
```cpp
int HardwareSerial::available(void)
{
  return (unsigned int)(_rx_buffer_head > _rx_buffer_tail ? _rx_buffer_head - _rx_buffer_tail : _rx_buffer_tail -_rx_buffer_head)
```
```cpp
int HardwareSerial::read(void)
...
    _rx_buffer_tail = (rx_buffer_index_t)(_rx_buffer_tail < SERIAL_RX_BUFFER_SIZE - 1 ? _rx_buffer_tail + 1 : 0);
```
```cpp
int HardwareSerial::availableForWrite(void)
  ...
  return head < tail ? tail - head - 1 : head - tail - 1;
```
```cpp
size_t HardwareSerial::write(uint8_t c)
  ...
  tx_buffer_index_t i = _tx_buffer_head < SERIAL_TX_BUFFER_SIZE -1 ? _tx_buffer_head + 1 : 0;
```

Note: the above code is completely untested.

If I get time, I will implement and test the changes and make a pull request referencing this issue.

matthijskooijman, 2018-11-13:
Did you check the generated assembly? I expect that, provided the buffer size is a power of 2, the compiler will turn the module into a single-cycle "and" operation (which is what is intended when writing this code).

When the buffer size is *not* a power of two, your analysis would certainly be correct and your proposal would likely be faster indeed (however, I think your proposal might be *slower* for power-of-two buffer sizes, I'm not sure if the compiler is smart enough to turn your code into a "and" instruction).

hattesen, 2018-11-13:
@matthijskooijman - no, I have not performed a disassembly of the generated code. But given that one of the two referenced articles [https://mekonik.wordpress.com/2009/03/02/modified-arduino-library-serial/] achieved a more-than-twofold increase in throughput, by replacing the modulo (%) operation with a bitwise-AND, when using a power-of-two size buffer, it seems that the compiler is not clever enough to optimize the modulo operation into a bitwise-and.

matthijskooijman, 2018-11-13:
The blogpost you linked is from 2009, so that will certain have been a significantly older compiler version. I would have expected the compiler from then to also perform the optimization, but perhaps I'm over-optimistic.

In any case, the current compiler might be smarter in this regard, so checking either performance or (preferably) generated assembly would be needed. If you want to have a stab, here's two links about Arduino assembly from the #arduino IRC bot:  http://rcarduino.blogspot.nl/2012/09/how-to-view-arduino-assembly.html and also http://www.gammon.com.au/tips#info1

hattesen, 2018-11-13:
@matthijskooijman and I believe you are indeed right.

The assembly of one of the modulo operations seem to be converted into a bitwise AND with the buffer size-1 (in this case 0x3F (63):

```
    _rx_buffer_tail = (rx_buffer_index_t)(_rx_buffer_tail + 1) % SERIAL_RX_BUFFER_SIZE;
 150:	92 8d       	ldd	r25, Z+26	; 0x1a
 152:	9f 5f       	subi	r25, 0xFF	; 255
 154:	9f 73       	andi	r25, 0x3F	; 63   <<<<<<<<<<<<<<<<<<<<<< bitwise AND 
 156:	92 8f       	std	Z+26, r25	; 0x1a
    return c;
 158:	90 e0       	ldi	r25, 0x00	; 0
 15a:	08 95       	ret
```

My bad for not checking that to begin with.
I still believe, however, that the source should not perform an order of magnitude worse, if someone chooses to use a non-power-of-two buffer (say, 10 bytes). That is a side-effect that you wouldn't expect and would be difficult to debug – especially when a source fix is easy (my modifications were maintaining source lines with no other changes). With a Macro, a bitwise AND could be used for power-of-two sizes, and a test/increment-or-reset if not power-of-two.

matthijskooijman, 2018-11-14:
Thanks for checking.

Could you also see what code is generated with your changes and a power-of-2 buffer size? Ideally, the compiler realizes that it can just do an "and" rather than a check and branch, but I'm afraid it won't see that :-)

aweatherguy, 2018-11-14:
I've been looking at this recently too. Here's the solution I'm considering right now. If fixes the problem with odd buffer sizes w/o any penalty with nice buffer sizes.

First, define this macro to detect whether the buffer size is power of two:

```
#define IS_PWR_TWO(x) ( ( (x) & ((x)-1) ) == 0 )
```

Then, to increment a buffer pointer with wrap-around, do this. 

```
#if IS_PWR_TWO(SERIAL_TX_BUFFER_SIZE)
    tail = (tail + 1) & (SERIAL_TX_BUFFER_SIZE - 1);
#else
    tail = tail + 1;
    if (tail >= SERIAL_TX_BUFFER_SIZE) tail = 0;
#endif
```

aweatherguy, 2018-11-14:
P.S. Yes - the compiler will do an "and" if you "modulo" with a power of two. The proposed solution above explicitly uses "and" to make it more obvious.

lxrobotics, 2019-09-17:
Thank you all for participating in this discussion :smiley: Since the compiler generates efficient instructions in cases where the buffer size is a power of two AND the size of the serial tx buffer within this core [are in fact a power of two](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/HardwareSerial.h#L44:L53) I'd say the advantage of creating arcane implementations for serial tx buffer sizes != a power of two is pretty small. On the other hand, the impact on code readability and maintainability is huge which is why I am closing this issue.

closed 2019-09-17 by lxrobotics