arduino/ArduinoCore-avr #77: Stream::find et al uses `char *` instead of `const char *`

majenkotech, 2019-03-05:
Character array parameters passed to functions, unless there is a good reason for it not to (such as if the function or a child function modifies the char array) should be set to `const char *` instead of `char *`. 

For example:

    bool find(char *target);

should be:

    bool find(const char *target);

The only time `char *` should be used is if the character array is to be modified.  This problem causes a warning "`ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]`" when using such things as `foo.find("bar");`

As a secondary point `find()` et al should really be expanded to support `const __FlashStringHelper` so that the `F()` macro can be used.

per1234, 2019-03-05:
Const correctness of the `Stream::find*` functions has been fixed by https://github.com/arduino/ArduinoCore-API/commit/1ebf4cd123b5c86d68607865fdc8cbf7fbca8108

There is an ongoing project ([Project Chainsaw](https://groups.google.com/a/arduino.cc/d/msg/developers/oyt_c5wePg8/_CWGyAG6CQAJ)) to move all the non-architecture specific core library code to its own repository ([ArduinoCore-API](https://github.com/arduino/ArduinoCore-API)) and share that code between all the official Arduino hardware cores (and hopefully 3rd party cores as well). ArduinoCore-API is already being used by [Arduino megaAVR Boards](https://github.com/arduino/ArduinoCore-megaavr)(though there hasn't been a release since https://github.com/arduino/ArduinoCore-API/commit/1ebf4cd123b5c86d68607865fdc8cbf7fbca8108 so the current release of megaAVR Boards still has the non-const issue with `Stream::find*`). For this reason, it probably makes the most sense to only make changes to that code in ArduinoCore-API repository and put the effort towards implementing ArduinoCore-API in all the official cores instead of towards porting changes from there to all the duplicate code in the current cores.

As for your request that the Stream find functions support `__FlashStringHelper`: that has not been implemented.

majenkotech, 2019-03-11:
That's awesome. I'll see about the possibility of integrating that repo into the chipKIT core.

I'll close this one now, and maybe open a fresh one on the core for adding `__FlashStringHelper` to the other repo.

closed 2019-03-11 by majenkotech