arduino/ArduinoCore-avr #86: [Wire] possibly missing break statements

wmarkow, 2019-05-07:
I have noticed those two places where possiblly a **break;** is missing:

https://github.com/arduino/ArduinoCore-avr/blob/5755ddea49fa69d6c505c772ebee5af5078e2ebf/libraries/Wire/src/utility/twi.c#L444-L448

https://github.com/arduino/ArduinoCore-avr/blob/5755ddea49fa69d6c505c772ebee5af5078e2ebf/libraries/Wire/src/utility/twi.c#L518-L534

Is this done in purpose or have I just found a bug?

matthijskooijman, 2019-05-07:
AFAIU these are on purpose. The idea is, I think, that the situation after a SLA (slave address) byte and after a data byte is largely the same. When a master is receiving data (the first case, Master Receiver), a data byte must be stored but after that the decision whether to continue with another data byte is the same as the decision whether to continue with the first byte after the address byte.

Similarly, for the Slave Transmitter case, after the address ack the buffer must be prepared, but after that sending the first byte works the same as sending subsequent bytes.

A comment to at least indicate the intentional fallthrough would have been appropriate, though.

wmarkow, 2019-05-07:
I think I understand. So for example in the Master Receiver case, **TW_MR_DATA_ACK** and **TW_MR_DATA_ACK** are almost the same, but for **TW_MR_DATA_ACK** a small additional work (copy incoming byte to internal buffer) must be done. To avoid the code duplication it goes to the next **case**.

It could be refactored to something like this:
```
// Master Receiver
case TW_MR_DATA_ACK: // data received, ack sent
case TW_MR_SLA_ACK:  // address sent, ack received
  if(TW_STATUS == TW_MR_DATA_ACK) {
    // put byte into buffer
    twi_masterBuffer[twi_masterBufferIndex++] = TWDR;
  }
  // ack if more bytes are expected, otherwise nack
  if(twi_masterBufferIndex < twi_masterBufferLength){
    twi_reply(1);
  }else{
    twi_reply(0);
  }
  break;
```
however it looks maybe a bit odd. I do not know :)

matthijskooijman, 2019-05-08:
Yeah, that would also work, but I'm not sure if it's really better than a fallthrough with a comment.

lxrobotics, 2019-09-17:
The missing break statements are intentionally missing :smirk: The warnings about implicit fallthrough have been fixed in https://github.com/arduino/ArduinoCore-avr/commit/786e405d54ab2151df3725366db8794b66086587 .

closed 2019-09-17 by lxrobotics