arduino/ArduinoCore-avr #91: USBtinyISP programmer does not work with avrdude 6.3.0-arduino16

MCUdude, 2019-06-03:
I just stumbled across a strange bug. The USBtinyISP programmer does not work with the most recent version of avrdude

```
/Users/hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino16/bin/avrdude -C/Users/hans/Documents/Arduino/hardware/MicroCore/avr/avrdude.conf -v -pattiny13 -cusbtiny -B32 -e -Ulock:w:0x3f:m -Uhfuse:w:0xfb:m -Ulfuse:w:0x3A:m 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/hans/Documents/Arduino/hardware/MicroCore/avr/avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
         Setting bit clk period        : 32.0
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 020:012
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash         65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          2    0      0     0     0 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Setting SCK period to 32 usec
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

Error while burning bootloader.

```

I opened my library folder and removed the packages folder. After restarting Arduino IDE I'm now able to communicate with the target. I'm using Arduino IDE 1.8.9 on MacOS.
(Don't worry about the incorrect target error. It's intended)
```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/hans/Documents/Arduino/hardware/MicroCore/avr/avrdude.conf -v -pattiny13 -cusbtiny -B32 -e -Ulock:w:0x3f:m -Uhfuse:w:0xfb:m -Ulfuse:w:0x3A:m 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/hans/Documents/Arduino/hardware/MicroCore/avr/avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
         Setting bit clk period        : 32.0
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 020:012
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash         65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          2    0      0     0     0 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Setting SCK period to 32 usec
avrdude: AVR device initialized and ready to accept instructions

Error while burning bootloader.
Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e930c
avrdude: Expected signature for ATtiny13 is 1E 90 07
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.
```

@facchinm any idea what may be causing this?

facchinm, 2019-06-04:
Hi @MCUdude ,
version 6.3.0-arduino16 contains this patch (https://github.com/facchinm/avrdude/commit/a45d20332a63fb529b5e8c89c7fea2bdf914e809) which was also merged in avrdude master (http://svn.savannah.gnu.org/viewvc/avrdude/trunk/avrdude/usbtiny.c?r1=1412&r2=1429).
The binaries in the original thread are still there (not for OSX though), could you try testing the arduino14 build? URL is http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino14-i386-apple-darwin11.tar.bz2

MCUdude, 2019-06-04:
Great! I'll test it when I get home from work

MCUdude, 2019-06-04:
It does work with avrdude-6.3.0-arduino14 on macOS!

What was your patch trying to solve? The >64kB issue?

facchinm, 2019-06-04:
The problem reported was more or less the same you reported (see https://github.com/arduino/Arduino/issues/8198#issue-380959747) and arised from supporting TPI programming on the attiny. The patch was confirmed to work on Windows so now I'm a bit scared to revert it (and regress the win build).

MCUdude, 2019-06-04:
The ATtiny13 uses the standard ISP interface, not TPI.

I hope we're able to get the USBtinyISP to work again. It's my main (and favorite) programmer

facchinm, 2019-06-10:
I checked the whole code again and indeed there's a bug I introduced :slightly_frowning_face: 
Since TPI patchset, `pgm->program_enable()` can return:
* `0` in case of success of TPI function
* `> 0 (true)` in case of success of normal function
and I didn't notice this asymmetry.
I'm going to push a patch, release `arduino17` version of avrdude and report to Joerg to also fix in mainline.
@MCUdude before going live I'll post a compiled package for OSX here, if you could test in your setup I'll merge all the stuff :slightly_smiling_face:

facchinm, 2019-06-10:
Compiled avrdude for testing
[avrdude-6.3.0-arduino16-x86_64-darwin13.zip](https://github.com/arduino/ArduinoCore-avr/files/3272139/avrdude-6.3.0-arduino16-x86_64-darwin13.zip)

MCUdude, 2019-06-10:
> Compiled avrdude for testing
avrdude-6.3.0-arduino16-x86_64-darwin13.zip

This version works correctly!

facchinm, 2019-06-12:
@MCUdude great, I'm preparing a full rebuild then for the next megaAVR/AVR release

facchinm, 2019-06-14:
I added the new avrdude rebuilt for all the platforms to this PR https://github.com/arduino/Arduino/pull/8976 . Would be great if you could test it :wink:

MCUdude, 2019-08-08:
Totally forgot about this one!  Avrdude 6.3.0-arduino17 does work with USBtinyISP again.

closed 2019-08-08 by facchinm