arduino/ArduinoCore-avr #102: Differential ADC support

Harvie, 2019-09-02:
**ATMega2560, ATMega32u4** and ATTiny85 have **differential ADCs with variable gain** and mux on chip. Can you please add support for these or arduino HAL? so everybody can use this with same ease as we use digitalRead() and analogRead()?

This probably covers **Arduino Mega 2560, Arduino Leonardo** and possibly Arduino Pro Micro (not sure about pinout). Also it might be usefull to support ATTiny85, since i use arduino to program these, but i understand there's no official Arduino board carrying ATTiny85.

Here are some resources:

https://forum.arduino.cc/index.php?topic=239030.0
https://www.davidpilling.com/wiki/index.php/DiffADC

Thanks

lxrobotics, 2019-09-17:
Hi @Harvie :wave: Thank you very much for raising this issue. The strength of Arduino lies within the minimal API which it exposes. Therefore any addition to the standard Arduino API has to be carefully considered and must be ported to all available cores, should the introduction of an additional API call be decided. As of right now differential ADCs are not a priority which is why I am closing this issue.

closed 2019-09-17 by lxrobotics

Harvie, 2019-09-17:
[image]