arduino/ArduinoCore-avr #104: SPI support SDIO half-duplex

warriorfenixSM, 2019-09-11:
Hi!

I need your help to know if the library support 3-wire SPI and if yes. How?

I attacht the picture.

[image]

I need to control a Optical track Sensor.

[PAT9130EW - AN04 SPI ControlFlow V0.1.pdf](https://github.com/arduino/ArduinoCore-avr/files/3602784/PAT9130EW.-.AN04.SPI.ControlFlow.V0.1.pdf)
[PAT9130EW-TKMT_GDS-R1.0_12062017_21006AEN.pdf](https://github.com/arduino/ArduinoCore-avr/files/3602785/PAT9130EW-TKMT_GDS-R1.0_12062017_21006AEN.pdf)

warriorfenixSM, 2019-09-13:
Hi i try to use SPI.begin transaction but not works

[image]

cmaglie, 2019-09-17:
Please move the discussion to the arduino forum:

forum.arduino.cc

you will surely find more help over there.

closed 2019-09-17 by cmaglie