arduino/ArduinoCore-avr #108: Arduino Leonardo only working on Windows 10 pc, its not working on Linux

delta010, 2019-09-16:
when I compile a sketch in Arduino Leanardo using Linux it shows an error but using windows it works without any problem

error in Linux:
    avrdude: verification error, first mismatch at byte 0x0000 arduino leonardo

lxrobotics, 2019-09-18:
@delta010 are you still experiencing this issue? I don't think the issue is sketch related and since you've removed `ModemManager` everything should be good.

isfanazha, 2020-02-01:
thanks, it works in Ubuntu 18.04 when I removed `ModemManager`, what is the effect if I remove `ModemManager`?

aentinger, 2020-02-03:
Quoting from the software package page ...

> ModemManager is a DBus-activated daemon which controls mobile broadband (2G/3G/4G) devices and connections. ModemManager is able to prepare and configure a wide variety of modems and setup connections with them.

So if you are using a `ModemManager` controlled modem on your linux system then removing `ModemManager` will render the functionality to be broken. If you are experiencing this you can simply reinstall it via
```bash
sudo apt-get install ModemManager
```

I'm closing this issue since further feedback from the original issue author @delta010 seems unlikely at this point. Please reopen if you still experience an issue despite removing `ModemManager`.

closed 2020-02-03 by aentinger