arduino/ArduinoCore-avr #109: wrong processing of uint64

ricardomsmonteiro, 2019-09-16:
When manipulating uint64_t, variables declared in a class in a translation unit, the results of operations are wrong if there is an access to other same class variables.

3 files were used for the purpose of showing this in the same arduino sketch:
issue.ino:
``` c++
#include "Arduino.h"
#include "tu.h"

#define ISSUE_HAPPENS

void setup()
{
  Serial.begin(115200);
}

void loop()
{
  foo.baz = 2;

#if defined(ISSUE_HAPPENS)
  if( (foo.bar  & (1ULL << 9)) && (foo.bar  & (1ULL << 10)) )
    Serial.println("right");
  else
    Serial.println("wrong");
#else
  uint64_t a = 0x8000000010000602;
  uint64_t b = 0x8000000010000602;
  
  if( (a  & (1ULL << 9)) && (b  & (1ULL << 10)) )
    Serial.println("right");
  else
    Serial.println("wrong");
#endif
  
  delay(1000);
}
```

tu.cpp:
```c++
#include "tu.h"

Foo foo;
```
tu.h
```c++
#include <Arduino.h>

#pragma once

class Foo
{
public:
  uint64_t bar = 0x8000000010000602;

  uint8_t baz = 4;
};

extern Foo foo;
```

The output, when `#define ISSUE_HAPPENS` is uncommented, is `wrong`. When it is commented it is `right` as it is supposed to be.

This happens in arduino 1.8.8 but not in arduino 1.8.5.  Using a different tool, sloeber arduino plugin for eclipse, it happens when the toolchain has gcc version 5.4.0 but not when it has the version 4.9.2.

Alexivia, 2019-09-16:
I have faced this same issue on a different situation. If using the [Arduino 1.8.5](http://downloads.arduino.cc/arduino-1.8.5-linux64.tar.xz) the OP code works as expected, *i.e.*, no 64 bit operation error. However, if I change the code to `if( (a  & (1ULL << 32)) && (b  & (1ULL << 62)) )` it breaks again, which means that on avr-gcc 4.9.2 the issue may manifest only if the operands larger than 1<<32

Are there any ongoing efforts to solve this?

matthijskooijman, 2019-09-20:
The recently released 1.8.1 AVR core contains a newer gcc (7.something). Perhaps someone could test these cases with that compiler, see if this is still broken?