arduino/ArduinoCore-avr #111: Arduino Leonardo. Mouse. How to use 16-bit data for XY axes?

125401qwerty, 2019-09-16:
Hi.
I'm doing a mouse on arduino and I ran into a problem:
How to use 16-bit data for XY axes? The fact is that the 8-bit data type is not enough for the optical sensor and when I put large DPI values then the mouse cursor makes a jerk. This happens when the value on the axis exceeds 127 (-127) and the cursor goes to 0.
Apparently it's in the file Mouse.cpp. It is necessary to change the descriptor.
In general, how to increase the limit on the XY axes in the -127 to +127, At least up to 255 -255 or -1024 +1024?
Maybe someone already solved this problem.

lxrobotics, 2019-09-17:
@raycewest Did you find anything during your investigation? If yes, we'd love to hear about it, if not, I'll close this issue.

PaulStoffregen, 2019-09-17:
I believe this issue should be moved to the Mouse library repository.

https://github.com/arduino-libraries/Mouse

As I understand this, the issue seems to be a misunderstanding about integer truncation.  It's pretty easy to see how this sort of misunderstanding could happen by looking at the Mouse.move() documentation.

https://www.arduino.cc/reference/en/language/functions/usb/mouse/mousemove/

While it does say "Allowed data types: signed char", nowhere on that page is the actual numerical range explained to be -128 to +127.  Neither does this page link to any other reference that explains the significance of what "Allowed data types: signed char" really means.

Even if the page did explain that, the HID report descriptor says the numerical range is -127 to +127.  Using -128 should not be allowed.

The Mouse.move() reference does have an example program, but the example is overly complicated and pretty far from the simple novice-friendly style Arduino normally uses for examples.  It also limits the range to only 12, without any explanation I can see.

At the very least, the reference page should be updated to explicitly state the allowed numerical range is -127 to +127.

A better, much more Arduino-style approach might change Mouse.move() to take int inputs instead of signed char, and explicitly clamp the range to -127 to +127.  Then if a user unwittingly tries to call Mouse.move() with values outside the legal -127 to +127 range, they would be nicely limited to the maximum rather than giving the confusing result of the compiler passing only the low 8 bits.

lxrobotics, 2019-09-18:
Unfortunately I do not have the power to transfer it to the [Mouse](https://github.com/arduino-libraries/Mouse) repository. @SimonePDA What do you think about updating the Arduino reference page as described by Paul?

facchinm, 2019-09-18:
I agree with the solution proposed by Paul. 
@lxrobotics Unfortunately an issue cannot be automatically moved to a repo under another organization, so moving manually (reopening there, closing this) would be the way.

lxrobotics, 2019-09-18:
Closing this issue since it was manually transferred by me into the [Mouse](https://github.com/arduino-libraries/Mouse/issues/9) repository.

closed 2019-09-18 by lxrobotics

NightStrang6r, 2023-06-11:
Hello. I rewrote the Mouse.h library to support 16 bits. Check out the [repository](https://github.com/NightStrang6r/USBHostShieldMouseProxy)