arduino/ArduinoCore-avr #112: Arduino Micro USB Serial cannot receive exactly 64 bytes at once

kervinck, 2019-09-16:
This issue was first reported in my project here https://github.com/kervinck/gigatron-rom/issues/36 but I now believe the root cause might be in USBCore.cpp

Tested in Ardiuno IDE 1.8.1 and 1.8.6 (arduino-PR-6886-BUILD-734-macosx.zip)
BN: Arduino/Genuino Micro
VID: 2341
PID: 8037
SN: Upload any sketch to obtain it

The example sketch just receives lines of text and reports their length:
```
void setup() {
  Serial.begin(115200);
}

void loop() {
  static int lineLength = 0;
  if (Serial.available()) {
    char next = Serial.read();
    lineLength++;
    if (next == '\n' || next == '\r') {
      Serial.println(lineLength);
      lineLength = 0;
    }
  }
}
```

Test Python program to demonstrate that the issue is with data messages of exactly 64 bytes:
```
import serial
from time import sleep
tty = '/dev/tty.usbmodem1411' # Adapt for your own computer
ser = serial.Serial(tty, baudrate=115200, timeout=1)
sleep(1) # For Arduinos that reboot

for length in [10, 70, 65, 64, 10, 10]:
  line = (length-1)*'A' + '\n'
  print('> %s (%d bytes)' % (repr(line), len(line)))
  ser.write(line)
  reply = ser.readline()
  print('< %s %s' % (repr(reply), 'OK' if reply else 'TIMEOUT'))
```

Output shows that the Arduino Micro stops receiving data after having received the 64-byte "killer" line. It does see the 64-byte message itself and it is still able to send. But it doesn't see new data anymore.
```
> 'AAAAAAAAA\n' (10 bytes)
< '10\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (70 bytes)
< '70\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (65 bytes)
< '65\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (64 bytes)
< '64\r\n' OK
> 'AAAAAAAAA\n' (10 bytes)
< '' TIMEOUT
> 'AAAAAAAAA\n' (10 bytes)
< '' TIMEOUT
```

Expected output is what the Arduino Uno does:
```
> 'AAAAAAAAA\n' (10 bytes)
< '10\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (70 bytes)
< '70\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (65 bytes)
< '65\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (64 bytes)
< '64\r\n' OK
> 'AAAAAAAAA\n' (10 bytes)
< '10\r\n' OK
> 'AAAAAAAAA\n' (10 bytes)
< '10\r\n' OK
```

Somewhat related to https://github.com/arduino/Arduino/issues/6669 ("Arduino Micro USB Serial cannot receive more than 255 data once") and https://github.com/arduino/Arduino/pull/6886 ("Handle receiving a ZLP in USB_Available"), however the proposed solution of that issue doesn't solve this.

I suspect the problem is caused by not following the datasheet's sequence, specifically:

> RXOUTI shall always be cleared before clearing FIFOCON

The way USB_Recv is written I don't think ReleaseRX will ever be called twice in a row. And that is exactly what gets me out of the hangup situation. In my own project I have integrated a workaround https://github.com/kervinck/gigatron-rom/commit/94d167dd755bdad22af6ae818f879f1d3e961e86 that seems to work by clearing FICOCON after Serial.read() and observing that UEBCLX has become 0. But I haven't considered every possible scenario (race conditions?).

----
I have some general remarks about USBCore.cpp, and this is _very_ subjective, so please take no offense. But having studied the code with an actual problem, the Arduino Micro hooked up and the datasheets at hand, I have the impression some more issues might be lurking here. It is not obvious (to me) that the code and the datasheet match.

-  Why not use RWAL in USB_Available instead of UEBCLX. Tracing its value in different scenarios, UEBCLX doesn't always seem quite the right indicator. It can sometimes momentarily read 0 in the middle of a transfer sequence. (Besides the question what to to with UEBCHX in case that ever becomes relevant)
-  What is the magic constant 0x6B in ReleaseRX? Why not address the desired bits explicitly with masking operations.
- USB_Recv() does some interesting things with its n and len, and the condition before ReleaseRX seems a bit suspect to me.

Unreal-Dan, 2023-01-30:
the problem is in the available() function it modulates by the buffer sizes to return the number of bytes available, so if you send exactly the buffer size it returns 64 % 64 as the amount available.

Serial::available needs to be fixed.

https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/HardwareSerial.cpp#L166

you can solve this by just sending one byte at a time but that shouldn't be the solution, we should be able to send any size frame of data and it should be reported as delivered.

matthijskooijman, 2023-01-31:
@Unreal-Dan I'm a bit confused. The report is about Arduino Micro USB serial, which is native CDC serial handled by https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/CDC.cpp, while you link to the HardwareSerial implementation that is used for UARTs only.

As for the fix, you suggest a `writeData` function that fixes things, but where should this function be applied? I cannot find a function by that name anywhere in this repository?

@tozz88 also wrote (before this issue was moved into the AVR core repo, I see):

> Has there been any new info on this issue? I am seeing similar behavior on the MKRZERO board. Using the above method I see failure in the following way. Results when attempting to send bytes from host
to MKRZERO over its USB COM port:

For reference, the code that handles MKRZERO lives at https://github.com/arduino/ArduinoCore-samd, so that is a different issue resulting from different code that should ideally be reported in that repository (if there is no open issue about it yet - I have not checked). It could be that the same mistake was made in that code too, but maybe it's a different problem with the same symptom, so better to track them separately.

Unreal-Dan, 2023-01-31:
I'm not super familiar with the arduino source code and I and just peeked quickly inside and assumed I knew the answer, I realize now that there are different implementations for available() 

Whatever the issue is, sending exactly 64 bytes is problematic on a trinket m0, the receiver never indicates any data is available and by simply sending one byte ahead of time and sending the rest of the 63.  

Perhaps this information can help to solve the problem

matthijskooijman, 2023-02-01:
> Whatever the issue is, sending exactly 64 bytes is problematic when I send it to a trinket m0, the receiver (trinket) never indicates any data is available and by simply sending one byte ahead of time and sending the rest of the 63 afterward everything works well.

Oh, that's even yet another board. The trinket M0 is runs on the [Adafruit's version of the samd core](https://github.com/adafruit/ArduinoCore-samd.git), which I guess is derived from Arduino's samd core. Good to know that all three of these cores (AVR, SAMD and Adafruit SAMD) are affected, though *this* issue is strictly about the AVR version only (but I suspect that there might be some shared code that is the underlying cause).

> When I say it is a solution, I mean it is a solution for people looking to communicate with their arduino without the 64 byte issue getting in their way.

Ah, thanks for clarifying.

PaulStoffregen, 2024-02-02:
I ran this on Arduino Leonardo using AVR boards 1.8.6.  Could not reproduce the problem.  Prints this:

```
> 'AAAAAAAAA\n' (10 bytes)
< b'10\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (70 bytes)
< b'70\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (65 bytes)
< b'65\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (64 bytes)
< b'64\r\n' OK
> 'AAAAAAAAA\n' (10 bytes)
< b'10\r\n' OK
> 'AAAAAAAAA\n' (10 bytes)
< b'10\r\n' OK
```

I did have to modify the python script slightly to work with the python3 install on Ubuntu 22.04.

```python
import serial
from time import sleep
tty = '/dev/ttyACM0' # Adapt for your own computer
ser = serial.Serial(tty, baudrate=115200, timeout=1)
sleep(1) # For Arduinos that reboot

for length in [10, 70, 65, 64, 10, 10]:
  line = (length-1)*'A' + '\n'
  print('> %s (%d bytes)' % (repr(line), len(line)))
  ser.write(line.encode())
  reply = ser.readline()
  print('< %s %s' % (repr(reply), 'OK' if reply else 'TIMEOUT'))

```

PaulStoffregen, 2024-02-02:
Also tried on Arduino Zero using SAMD boards 1.8.14.  To use the native port I had to replace "Serial" with "SerialUSB", as "Serial" talks to the debug port.  This is the code I ran on Arduino Zero:

```c++
void setup() {
  SerialUSB.begin(115200);
}

void loop() {
  static int lineLength = 0;
  if (SerialUSB.available()) {
    char next = SerialUSB.read();
    lineLength++;
    if (next == '\n' || next == '\r') {
      SerialUSB.println(lineLength);
      lineLength = 0;
    }
  }
}
```

Also could not reproduce the problem, running the python script on Ubuntu 22.04.  Got this again:

```
> 'AAAAAAAAA\n' (10 bytes)
< b'10\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (70 bytes)
< b'70\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (65 bytes)
< b'65\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (64 bytes)
< b'64\r\n' OK
> 'AAAAAAAAA\n' (10 bytes)
< b'10\r\n' OK
> 'AAAAAAAAA\n' (10 bytes)
< b'10\r\n' OK
```

And FWIW, also tested on Teensy 4.1, Teensy 3.2, Teensy LC, Teensy++ 2.0 and Teensy 2.0.  Could not reproduce the problem on any of those either.

PaulStoffregen, 2024-02-02:
One last test, MKR1000 with the original code (not "SerialUSB").  Also works.

```
> 'AAAAAAAAA\n' (10 bytes)
< b'10\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (70 bytes)
< b'70\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (65 bytes)
< b'65\r\n' OK
> 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\n' (64 bytes)
< b'64\r\n' OK
> 'AAAAAAAAA\n' (10 bytes)
< b'10\r\n' OK
> 'AAAAAAAAA\n' (10 bytes)
< b'10\r\n' OK
```

PaulStoffregen, 2024-02-02:
Unless someone can confirm the problem happen with Windows or MacOS, seems likely this issue is no longer an issue.