arduino/ArduinoCore-avr #113: Adding SPI programming on a Raspberry Pi (from the IDE)

lewispg228, 2019-09-16:
I've been working on a [stand-alone AVR programming solution](https://learn.sparkfun.com/tutorials/raspberry-pi-stand-alone-programmer) for SparkFun over the past couple years. Currently, we use Python (watching a GPIO) to engage a bash file that calls avrdude.  We have found that the SPI hardware built into the raspberry pi is very reliable and it would be awesome to be able to also develop within the arduino IDE (on a raspi), and upload via SPI.

Would this be possible? Is it a crazy request?

lxrobotics, 2019-09-17:
Closing this issue since a solution is available as mentioned in above post by @facchinm ( https://github.com/arduino/ArduinoCore-avr/issues/113#issuecomment-406189258 ).

per1234, 2019-09-17:
@lxrobotics this proposal is about adding a linuxspi programmer option to the **Tools > Programmer** menu in the Arduino IDE so that this programmer can be used from inside the Arduino IDE. The comment you referenced is not at all a solution for that.

I understand completely if the decision is that Arduino does not want to add official support for this programmer, but, if so, that should be stated clearly as the reason for closing the issue.

I've been able to implement this, but it was blocked by a bug in avrdude. Although that that patch was merged long ago, there was no release of a new version of avrdude since then. I believe it might be in avrdude 6.3.0-arduino17 that comes with the newly released Arduino AVR Boards 1.8.1, though I haven't checked that yet since no versioning system is used in Arduino's avrdude development process.

lxrobotics, 2019-09-17:
Sorry, I've apparently misread and will reopen this issue. In any case, it's wrongly placed here then and should be moved into the Arduino IDE repository.