arduino/ArduinoCore-avr #129: stk500v2 bootloader corrupts data when reading EEPROM

mlasevich, 2019-09-16:
When reading EEPROM via avrdude using wiring (via bootloader) - results are corrupted. Every other 8 bytes are lost. Writing to EEPROM works, but, as expected, validation fails as what is written does not match corrupted version that is read back.

You can reproduce on Mega2560(and probably others using this bootloader) using following sketch to initialize EEPROM to known state:

```c++
#include <EEPROM.h>

void eepromDump() {
  char buff[8];
  int counter = 0;
  byte data = 0;
  Serial.write("EEPROM contents:\n");
  for (int i = 0; i < 4096 ; i++) {
    counter = counter % 16;
    if (counter == 0) {
      snprintf(buff, 10, "%07X", i);
      if (i != 0) {
        Serial.println();
      }
      Serial.print(buff);
    }
    data =  EEPROM.read(i);
    snprintf(buff, 4, " %02X", data);
    Serial.print(buff);
    counter ++;
  }
  Serial.println();
}

void eepromErase(int start){
  Serial.println("Erasing EEPROM");
  for (int addr=start; addr <4096; addr++){ EEPROM.update(addr, 0x00); }
}

void setup() {
  Serial.begin(115200L);
  const char data[63]="0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ";
  EEPROM.put(0, data);
  eepromErase(64);
  eepromDump();
}

void loop() {
}
```

After that if executing a read via a programmer you get expected correct output:
```
$ avrdude -c dragon_isp -P usb -p atmega2560 -Ueeprom:r:eeprom.bin:r
$ hexdump -C eeprom.bin
00000000  30 31 32 33 34 35 36 37  38 39 61 62 63 64 65 66  |0123456789abcdef|
00000010  67 68 69 6a 6b 6c 6d 6e  6f 70 71 72 73 74 75 76  |ghijklmnopqrstuv|
00000020  77 78 79 7a 41 42 43 44  45 46 47 48 49 4a 4b 4c  |wxyzABCDEFGHIJKL|
00000030  4d 4e 4f 50 51 52 53 54  55 56 57 58 59 5a 00 ff  |MNOPQRSTUVWXYZ..|
00000040  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00001000
```

But using the bootloader for same thing you get corrupted output:

```
$ avrdude -c wiring -P /dev/cu.usbmodem1411 -p atmega2560 -Ueeprom:r:eeprom2.bin:r
$ hexdump -C eeprom2.bin 
00000000  30 31 32 33 34 35 36 37  67 68 69 6a 6b 6c 6d 6e  |01234567ghijklmn|
00000010  77 78 79 7a 41 42 43 44  4d 4e 4f 50 51 52 53 54  |wxyzABCDMNOPQRST|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000800  30 31 32 33 34 35 36 37  67 68 69 6a 6b 6c 6d 6e  |01234567ghijklmn|
00000810  77 78 79 7a 41 42 43 44  4d 4e 4f 50 51 52 53 54  |wxyzABCDMNOPQRST|
00000820  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00001000
```

mcuee, 2023-06-09:
PR #24 may have fixed this issue.

For those who want to use an Arduino Core with PR #24 , you can use this one.
https://github.com/MCUdude/MegaCore

It has incorported #24 fix here.
https://github.com/MCUdude/MegaCore/commit/eba4f4a827aa3597b8e65144caee83f69724f280

mcuee, 2023-06-10:
The zip file here contains this EEPROM fix and a few other fixes.
* https://github.com/avrdudes/avrdude/issues/1379#issuecomment-1585533005