arduino/ArduinoCore-avr #132: wiring.c select which timer to use for millis() and other functions

sbasmanov, 2019-09-16:
Would be great to be able to define which timer to use for millis/micros/delay and may be others, where timer0 currently used. For example timer2 (where available) can be used instead.
This will be useful in case both timers 0 and 1 have to be clocked from external source to count pulses from two independent sources. Something like #define CORE_USE_TIMER2

rtek1000, 2022-02-02:
Hello @per1234,

Sorry, I don't understand, has this improvement already been implemented in the current cores?

If it wasn't implemented, and there is already an example for it, why wasn't it implemented in the current cores?

Is there something missing for this to work?

Thank you.

per1234, 2022-02-02:
>has this improvement already been implemented in the current cores?

No. This issue is used to track the feature request. It will be closed when the feature has been either implemented or declined by the developers of the project.