arduino/ArduinoCore-avr #144: Arduino Uno AttachInterrupt wrong trigger

ygnd1, 2019-09-16:
Hi,
I'm trying to detect two hall sensors with AttachInterrupt event using Arduino Uno module.

The code attached

void flow(){
     flow_frequency++;
}
void flow_1(){
     flow_frequency_1++;
}
void setup(){
   Serial.begin(9600);  
   pinMode(2, INPUT);
   attachInterrupt(digitalPinToInterrupt(2), flow, RISING); 
   pinMode(3, INPUT);
   attachInterrupt(digitalPinToInterrupt(3), flow_1, RISING);  
   sei();
 }

When the input 2 is rising the event "flow" trigger and also event "flow_1" is trigger which cause to flow_frequency_1 parameter to increment and it is worng.

When input 3 is rising event "flow_1" trigger and event "flow" not trigger which is ok.

Please advise ?

Thank you

Yalon

lxrobotics, 2019-09-17:
@ygnd1 can you confirm wether or not this still is an issue for you?

cmaglie, 2019-09-18:
This issue is quite old, I guess @ygnd1 solved the issue in a way or another in the meantime.

BTW the forum is more suited for this kind of questions: https://forum.arduino.cc/

closed 2019-09-18 by cmaglie