arduino/ArduinoCore-avr #149: pinMode() does not check valid pin number

mikaelpatel, 2019-09-16:
The current [pinMode](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/wiring_digital.c#L29)() implementation performs a digitalPinToPort(pin) and then checks the port for NOT_A_PORT but [digitalPinToPort](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Arduino.h#L169)() is a program memory table read. It is possible to give an illegal pin number. 

Possible correction is:

```
void pinMode(uint8_t pin, uint8_t mode)
{
    if (pin >= NUM_DIGITAL_PINS) return;
    uint8_t bit = digitalPinToBitMask(pin);
    uint8_t port = digitalPinToPort(pin);
    volatile uint8_t *reg, *out;
    ...
}
```

Replacing the port check with a pin number check (first). 

Cheers!

neu-rah, 2019-09-16:
https://github.com/arduino/ArduinoCore-avr/pull/2

lxrobotics, 2019-09-17:
@mikaelpatel Good catch :+1: Could create a small PR for this?

cousteaulecommandant, 2020-02-25:
> @mikaelpatel Good catch +1 Could create a small PR for this?

Back in the day I opened arduino/Arduino#4649, but it was closed after Chainsaw and never reopened.
Now there is #302 targeting the same issue (not sure how different it is or which approach is better, although I do remember my approach was more of a hack than a proper fix).