arduino/ArduinoCore-avr #150: ProMini: Reboot loop when using watchdog

tuxedo0801, 2019-09-16:
Hi there,

i have the following problem:

I have an arduino sketch which accept some configuration-details over the serial connection. After applying some configuration, a "reboot" is required. On my leonardo and micro boards, this works quite well by using the watchdog to reset the board:

``` c++
    wdt_enable( WDTO_1S ); 
    while(1) {}
```

But with an arduino pro mini  at 3,3V/8Mhz the board loops in a restart-loop, forever.
It seems that this issue is already known on the net:

https://andreasrohner.at/posts/Electronics/How-to-make-the-Watchdog-Timer-work-on-an-Arduino-Pro-Mini-by-replacing-the-bootloader/

I know that there are othr techniques to "reboot" an arduino, but the WDT seems to be the only real hardware reboot, triggered by software.

Of course I can use the optiboot bootloader to overcome this issue. But this adds some additional complexity and arduino ide tweaking-skills, which I would like to avoid (there are other people using my sketches/libraries which won't be able to tweak the arduino ide properly).

Would it be possible to fix this in arduino ide direclty? For example by switching to optiboot by default or fix the "bug" in the current used bootloader?!

br,
 Alex

kai-morich, 2019-11-30:
Instead of switching to alternative bootloader, I used a free timer + pin to create a custom watchdog:

```
class Watchdog {
  public:
  
  void init() {
    TCCR1A = _BV(COM1A0);// OC1A toggle
    TCCR1C = _BV(FOC1A); // toggle manually, else would start initially LOW
     TCNT1 = 0;
    TCCR1B = _BV(CS12);  // start timer with clk/256 => overflow after 2sec @ 8MHz
    pinMode(9, OUTPUT); // = OC1A
  }
  
  void reset() {
    TCNT1 = 0;
  }
};
```
and connect OC1A (pin 9) with diode --|<|--- to RST

Dancofra, 2020-08-31:
fix the WatchDog problem with the Arduino Promini loaded the optiboot_atmega328.hex to the Arduino Promini,

loaded the new project from the Arduino IDE selected the Arduino UNO and the COM port

optiboot-master:  https://github.com/Optiboot/optiboot

works very well ---- it is just loading the programs as if it were for an Arduino UNO

#include <avr/wdt.h>

void setup(){
  Serial.begin(9600);
  Serial.println("Setup started :");
  // make a delay before enable WDT 
  // this delay help to complete all initial tasks
  delay(2000);
  wdt_enable(WDTO_4S);
}

void loop(){
  Serial.println("LOOP started ! ");
  for(int i=0; i<=5; i++){
    Serial.print("Loop : ");
    Serial.print(i);
    Serial.println();
    delay(1000);
    wdt_reset();
  }
  //infinity loop to hang MCU
  while(1){}
}

chomchom69420, 2023-04-21:
Hello, raising this issue after a long time. 

I am working on an assignment where I have to force restart an atmega328 using the wdt. I am clearing the WDIE bit to stop the interrupt and directly move into system reset. 
Here is the code for the function:

`void wdt_force_restart()
{
    //disables interrupt
    SREG &= ~(1<<I_SREG);

    //Enabling change
    WDTCSR |= (1<<WDCE);

    //Clearing WDIE
    WDTCSR &= ~(1<<WDIE);

    //Wait for reset
    while(1);
}`

This should reset my atmega and it should print out the initialization lines as given here:

`int main(void)
{
    /* initialize the hardware and device state */
    config_init();
    vpd_init();
    rtc_init();
    log_init();
    alarm_init();
    uart_init();
    led_init();
    temp_init();
    wdt_init();
    tempfsm_init();

    unsigned char connected = 0;  /* 1 if socket is connected, otherwise, 0 */

    /* print the banner to the debug port */
    uart_writestr("SER486 Project 3 - Integration\r\n");
    uart_writestr( STRINGIZE_VALUE(YOURNAME) );
    uart_writestr("\r\n");`

The above code is from the main.c file.

I am using the simulavr platform to simulate the code. Whenever I request a reset, the device is getting stuck in the infinite while() loop and not resetting after 2s (which is the time I have setup as below):

`void wdt_init()
{
    //Clear WDRF in MCUSR
    //MCUSR &= ~(1<<WDRF);

    //Setting the pre-scalar for timeout period of 2s
    //WDTCSR |= (1<<WDP0) | (1<<WDP1) | (1<<WDP2);

    //Interrupt and system reset mode
    WDTCSR |= (1<<WDIE) | (1<<WDE);

    //Initizaize global interrupts
    SREG |= (1<<I_SREG);
}`

Could anyone suggest a fix for this issue? Really appreciate it. :)

Is there a way of changing the bootloader in simulavr, as I can see above apparently switching to Optiboot helps. I don't know much about simulavr and I am not sure what bootloader it uses by default.

per1234, 2023-04-21:
Hi @chomchom69420. Thanks for your interest in this open source project. This issue thread is only to be used for discussion directly relevant to the resolution of the reported bug. The discussion of your project is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

mirh, 2023-05-14:
> we'll not be producing Pro/Pro mini anymore so there won't be any bootloader update for these boards (following the usual "don't update the bootloader if it's not being flashed in production").

While reading on #308, an incredible idea came to my mind. 
Why don't you make available an updated bootloader, ***but*** it's a separate option in the IDE just like you already have on the Nano?
Make it even non-default if you really want to be conservative, make the name scary or stick a big BETA label if your years-long testing still hasn't gone through, but this way you could make everyone happy.

rtek1000, 2023-06-08:
Hello, I fell back into this bad Bootloader hole, on a Pro Mini.

The board is a bit difficult to access, so updating the bootloader via ISP is a bit tricky now.

--------

I've seen something about making a sketch that can update the bootloader, but it seems to be something obscure that not many people have tried:

Ref.: [Update bootloader using specialy crafted Arduino sketch](https://github.com/Optiboot/optiboot/issues/264)

--------

But I managed to get out of the state of flashing (forever) the LED,
- (Note: This does not solve the problem of the WDT, but it works again without the WDT)

> I noticed that when powering the board, it takes a long time for the LED (pin 13) to start blinking, but after the LED is blinking, when RESET is pressed, the LED blinks quickly again, so the bootloader does not have time to rewrite the sketch

Steps:

1- Activate "compilation" and "upload" LOGs (verbose output) in preferences (restart IDE)

2- Completely turn off the Pro Mini power supply, including the USB Serial adapter

3- Press and hold RESET (keep pressing RESET until the middle of step 6) on the Pro Mini board (this is fundamental to work)

4- Connect the USB Serial adapter (which will also power the Pro Mini board)

5- Check and select the port again in the Arduino IDE (may happen if the system re-enumerates the port)

6- Upload a sketch without WDT [without wdt_enable()] to the Pro Mini board, after compiling and reaching the Upload part, where orange letters appear, **release** the Pro Mini RESET button, this should give the necessary time for the bootloader to work

(Tested with: Arduino IDE 1.8.19, Linux OS and Red FTDI Type USB Adapter)