arduino/ArduinoCore-avr #171: Wire.requestFrom returns the number of bytes requested, regardless of what is sent

nickgammon, 2019-09-16:
Example slave:

```
#include <Wire.h>
void setup ()
{
Wire.begin (42);
Wire.onRequest (requestEvent); // interrupt handler for when data is wanted
} // end of setup
void requestEvent ()
{
Wire.write ("ABCDE", 5);
} // end of receiveEvent
void loop() { }
```

This slave always returns 5 bytes.

- - -

Example master:

```
#include <Wire.h>
void setup ()
{
Wire.begin ();
Serial.begin (115200); // start serial for output
} // end of setup
size_t count;
void loop()
{
count++;
if (count > 32)
while (true) {}
byte returnedCount = Wire.requestFrom (42, count);
Serial.println (int (returnedCount));
} // end of loop
```

This master requests from 1 to 32 bytes from the slave.

- - -

Results:

```

1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
```

The master always gets what it **requested**, not 5, which is what the slave **sent**.

- - -

The reference page http://arduino.cc/en/Reference/WireRequestFrom says:

> Returns
> 
> byte : the number of bytes returned from the slave device 

It is not returning the number of bytes returned from the slave device, it is returning the number of bytes **requested** from the slave device.

Also relevant: https://github.com/arduino/Arduino/issues/1311

- - -

I doubt it is even possible for the master to obtain the number of bytes sent by the slave, as the **master** controls the clock, and thus the slave cannot send a "stop condition" when there is no more to send.

Jeremywinst, 2020-03-31:
100% WORKS. check this out
https://dronebotworkshop.com/i2c-arduino-arduino/

SherifMEldeeb, 2021-04-27:
> fixed with 98d0a72, reopen if needed

Does it work as expected now or the same behavior as the OP stated still exist because I am having the same issue here.

pat1, 2022-02-01:
> fixed with 98d0a72, reopen if needed

the commit is for sam only