arduino/ArduinoCore-avr #177: HardwareSerial atomicity issues

matthijskooijman, 2019-09-16:
I just realized that there might be an atomicity issue in HardwareSerial. So far, we've assumed that the RX buffer is filled by an ISR and emptied by the main loop and that the TX buffer is filled by the main loop and emptied by an ISR (in other words, the head and tail pointers are only written in one "thread" and read by the other). However, when writing to or reading from the serial port from inside an (other) ISR violates this assumption.

Recently, HardwareSerial was modified to not lock up when writing from inside an ISR, but right now bytes written inside an ISR could still be dropped (or bytes read could be read twice). Note that it can still happen that serial printing blocks for a while when the buffer is full, which is typically a bad idea inside an ISR.

The problem is that updating the head and tail pointers is not atomic - it's read - increment - write. If, inside the write function an interrupt triggers between the read and the write that also calls write, the change of the head pointer that happens inside the ISR will be overwritten by the main thread - effectively dropping the byte.

To fix this, the increment should happen with (all) interrupts disabled, incurring a small overhead and interrupt latency.

Do we think this is a problem? Or is serial printing from inside an ISR considered bad practice and to be used for debugging only and is dropping a byte here or there acceptable?

chrysophylax, 2021-05-21:
Has this been fixed?

JeroenIoT, 2022-10-05:
> @matthijskooijman Your observation is spot on, there are two different albeit related issues, just as you state. For buffers > 256 the warning in the header file is an understatement, havoc will occur.
> 
> I encountered this problem in my own projects long ago and made a fix that I never came around to submitting. Will revisit the issue and propose the fix, building on the pattern you showed.

@egilkv - did you ever come to the point to propose the fix? I am having some issues with large size buffers (> 256) and would be pleased if you can share your fix. Thnx

SurfGargano, 2023-03-31:
Should not an issue with 32 bit mcu , if head and tail is 32-bit aligned

matthijskooijman, 2023-03-31:
> Should not an issue with 32 bit mcu , if head and tail is 32-bit aligned

I'm not sure if this is really true (though there might be special atomic increment instructions), but it is also completely not relevant to this particular issue (since this issue is in the ArduinoCore-avr repository, which is an 8-bit MCU).

SurfGargano, 2023-03-31:
The module is also used in platformio/arduino lib of stm32

matthijskooijman, 2023-03-31:
> The module is also used in platformio/arduino lib of stm32

Nope, STM32 Arduino cores (there's at least two as far as I know) come from a different repository with unrelated hardwareserial code (maybe they have a filename by the same name, but the code is completely independent). And again, *this* repo and issue is about AVR, not STM32. Anyway, this is all detracting from the subject of this issue, so let's not waste more words about it.

ZhaoSQ-CH, 2023-05-24:
//< If the buffer is too small, it will cause the program to crash. It is recommended to change it to a larger size, and change it to 128
#define SERIAL_RX_BUFFER_SIZE 128
#define SERIAL_TX_BUFFER_SIZE 128

ZhaoSQ-CH, 2023-05-24:
If I don't want to modify SERIAL_RX_BUFFER_SIZE/SERIAL_TX_BUFFER_SIZE in the HardwareSerial_h header file, where can I modify it, I am using platformIO (based on stm32 ).

SurfGargano, 2023-05-24:
You can define it in main.cpp prior #include hardwareserial.h or in platform.ini

matthijskooijman, 2023-05-24:
> You can define it in main.cpp prior #include hardwareserial.h 

That is a common misconception - the buffers are defined in (IIIRC) HardwareSerial.cpp, which does *not* see your define from main.cpp, so will just use the original. To make sure the defines end up there, you need to make sure to put them on the compiler commandline (which is not really supported by Arduino, I'm not sure about platformio, maybe that's done in `platform.ini`?). Another way could be to define it in any .h file that is (directly or indirectly) included by HardwareSerial.cpp or .h, but that should normally only include headers from within the core, so that's not much help.

ZhaoSQ-CH, 2023-05-27:
Thank you, I have succeeded, after the hardwareserial.h file is modified, then the compilation is successful. 

  

  
 
&mdash;