arduino/ArduinoCore-avr #179: caterina bootloader

fricklr, 2019-09-16:
problem with bootloader handover:
http://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&t=123862
+++ bootloaders/caterina/Caterina.c
@@ -151,9 +151,31 @@
        LEDPulse();
    }
-  // Disable the USB interface
-   USBCON &= ~(1 << USBE);
-   
-   // Disable the VBUS transition enable bit
-   USBCON &= ~(1 << VBUSTE);
-   
-   // Disable the VUSB pad
-   USBCON &= ~(1 << OTGPADE);
-   
-   // Freeze the USB clock
-   USBCON &= ~(1 << FRZCLK);
-   
-   // Disable USB pad regulator
-   UHWCON &= ~(1 << UVREGE);
-   
-   // Clear the IVBUS Transition Interrupt flag
-   USBINT &= ~(1 << VBUSTI);
-   
  +

cmaglie, 2019-09-17:
I've looked at the linked forum thread, and it seems that the issue was fixed in LUFA at the times, so I'm closing this one. Feel free to comment further if I'm wrong.

closed 2019-09-17 by cmaglie

teliot, 2021-09-29:
Ran into this issue when I disabled USB which made the MCU unresponsive once sei() was called as USB interrupts were triggering but not correctly handled.

Changing the Caterina boot loader would fix the issue for others in the future. I have no idea if this would require a matching change in the Arduino USB CDC code to correctly re-enable the USB endpoints.

156a157,161
> 	/* clear leftover USB settings from bootloader or application interrupts will fail with USB disabled*/
> 	USBCON = 0;
> 	USBINT &= ~(1 << VBUSTI);
> 	UHWCON &= ~(1 << UVREGE);
>