arduino/ArduinoCore-avr #192: memory improvement of the tone.cpp lib

RobTillaart, 2019-09-16:
To find the optimal prescaler a formula is repeatedly calculated (9 times). By replacing this expensive formula (long divisions) partly by a precalculated value the size of a sample sketch, and thus the tone lib was reduced by 88 bytes. 
Did not test the code but I assume it executes a tiny bit faster.

improved code snippet

```
void tone(uint8_t _pin, unsigned int frequency, unsigned long duration)
{
  uint8_t prescalarbits = 0b001;
  long toggle_count = 0;
  uint32_t ocr = 0;
  int8_t _timer;

  _timer = toneBegin(_pin);

  if (_timer >= 0)
  {
    uint32_t ocrRaw = F_CPU / frequency / 2;
   ....
```

   and replace F_CPU / frequency / 2 in the rest of the function with ocrRaw.

See forum - http://arduino.cc/forum/index.php/topic,142370.msg1069191.html#msg1069191 -

neu-rah, 2019-09-16:
here, for AVR timers, a function that selects prescaler to best match pwm frequency (not just tone)
https://github.com/neu-rah/Fielduino/blob/master/fielduino/avrTC.h#L239

if it helps

RobTillaart, 2020-05-03:
too old to be relevant anymore

closed 2020-05-03 by RobTillaart