arduino/ArduinoCore-avr #206: serialEvent does not work with Leonardo (USB CDC Serial) [imported]

cmaglie, 2019-09-16:
This is [Issue 1031](http://code.google.com/p/arduino/issues/detail?id=1031) moved from a Google Code project.
Added by 2012-09-03T19:30:50.000Z by [pilot...@gmail.com](http://code.google.com/u/107526621707702953137/).
Please review that bug for more context and additional comments, but update this bug.

Original labels: Type-Defect, Priority-High, Component-ATmega32U4
### Original description

**What steps will reproduce the problem?**
1. Load Examples ... 04. Communcations... SerialEvent example 
2. Add while (!Serial) ; after Serial.begin() in setup
3. Upload to a Leonardo board, type stuff in the serial monitor

**What is the expected output? What do you see instead?**

Expected: Get back what I typed
Actual: Got nothing back

**What version of the Arduino software are you using? On what operating**
**system?  Which Arduino board are you using?**

Arduino 1.0.1 on MacOS X 10.7.4 with a Leonardo R3 board

**Please provide any additional information below.**

I believe serialEvent is never called. I tried renaming it to serialEvent1 and serialEvent2 without any effect.

facchinm, 2019-09-17:
Fixed (or, better, documented) by https://github.com/arduino/Arduino/commit/e180b7037dcf303fb3ff27dc60fadcee5d7f3a09 .
Anyway, `serialEvent` is due to deprecation, so no fixes will be applied to code not working today.

closed 2019-09-17 by facchinm

PaulStoffregen, 2019-09-17:
If serialEvent is depreciated, maybe users could be advised not to use it on these pages?

https://www.arduino.cc/en/Tutorial/SerialEvent

https://www.arduino.cc/en/Reference.SerialEvent

JoakimCh, 2020-11-10:
@PaulStoffregen Indeed! I happened to spend (waste) several hours now just to figure this out... I see no logical reason for it to work with some Arduinos and not this one, hence it not being implemented was very confusing. But I guess this is to be expected with the low quality of most of the standard Arduino libraries...

PaulStoffregen, 2020-11-11:
Not sure I agree with your quality assessment.  But I do believe designing public APIs with weak linking of function names is a poor practice.

Looks like Arduino did update the primary documentation.

[image]

PaulStoffregen, 2020-11-11:
However, this old tutorial page could really use an update to mention which boards do and don't support serialEvent.

https://www.arduino.cc/en/Tutorial/BuiltInExamples/SerialEvent

If it really is going to be depreciated, might be good to update the reference and tutorial to recommend against using serialEvent?

@per1234 - How do we request documentation updates these days?

per1234, 2020-11-11:
@PaulStoffregen the `arduino/Arduino` repository's issue tracker seems to serve as the catch-all for issues for which there isn't a more appropriate public issue tracker.

If someone submits specific recommendations for improving/correcting the Tutorials pages (with the exception of the "Hacking" section as I don't have access to the source of that content), I'll submit a PR to implement those changes if I'm in agreement and am able to personally verify the information is accurate.

I'm not the one to make decisions regarding deprecation though.

PaulStoffregen, 2020-11-12:
I would recommend 2 specific changes to that tutorial page.

1: Add "serialEvent() is not compatible with the Esplora, Leonardo, or Micro" somewhere on the page.

2: In the first sentence, change "SerialEvent()" to "serialEvent()".  It should not be capitalized.

per1234, 2020-11-16:
Thanks for your recommendations @PaulStoffregen!

>1: Add "serialEvent() is not compatible with the Esplora, Leonardo, or Micro" somewhere on the page.

I did a survey of the situation and decided that the best approach is to completely document `serialEvent()` support for all official boards in a single place, then link to that from the tutorial. This will make it easier/more likely to keep the information updated.

As the first step, I have submitted a PR to fully document the situation on the Arduino Language Reference's `serialEvent()` page:
https://github.com/arduino/reference-en/pull/800
I would welcome any reviews or feedback on that PR.

Once I have received an approval on https://github.com/arduino/reference-en/pull/800, I will then submit a PR to add a link to that information to both the serialEvent() tutorial page and the comment in the [built-in "SerialEvent" example sketch](https://github.com/arduino/arduino-examples/blob/main/examples/04.Communication/SerialEvent/SerialEvent.ino) associated with that tutorial.

>2: In the first sentence, change "SerialEvent()" to "serialEvent()". It should not be capitalized.

I have submitted a PR to correct this.