arduino/ArduinoCore-avr #259: Optimization fix: -freorder-blocks compiler option

xxxajk, 2019-09-16:
Adding -freorder-blocks to the compile options fixes a random crash problem I was having with a large project. It does not increase the code size all that much when enabled... on average 200 to 400 bytes of flash are consumed on a large project.

It is obvious that -Os by its self can in some cases generate bad code, and this is an acceptable tradeoff. i.e. works, instead of banging your head on the desk wasting time trying to find out why your code is broken, when it isn't.

lxrobotics, 2019-09-17:
@xxxajk Can you confirm if this issue is still relevant?

xxxajk, 2019-09-17:
Haven't had a chance to confirm this recently, but the issue was still present last time I encountered it. 
It seems to depend on how complex the function and data is.

lxrobotics, 2019-09-18:
I've just checked with documentation and it seems that `-freorder-blocks` is automatically enabled at  `-O, -O2, -O3, -Os`. Explicitly adding the compile time flag should not change anything.