arduino/ArduinoCore-avr #266: Arduino fails to optimize out unreachable Serial library code

etherfi, 2019-09-16:
Many libraries use Serial output in debugging routines, and often times the serial interface is used for any number of other purposes.

Sometimes though, in a finished device, the serial interface is not used at all. In these instances, those serial output routines are never called, yet the Serial library wastes precious Flash space and SRAM.

Here's an empty Arduino project, and its compilation output:

```
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

Sketch uses 450 bytes (1%) of program storage space. Maximum is 32,256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
```

Now the same code again with a simple, but unused function added:

```
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

void unusedFunction() {
  Serial.println();
}

Sketch uses 1,388 bytes (4%) of program storage space. Maximum is 32,256 bytes.
Global variables use 182 bytes (8%) of dynamic memory, leaving 1,866 bytes for local variables. Maximum is 2,048 bytes.
```

As you can see, 173 bytes or SRAM and 938 bytes of Flash are now used that should have been available to the programmer. The SRAM is of the most concern... That's almost 17% of the SRAM available on 1k chips!

And to trigger this bug, all you have to do is have a stray Serial library reference in and file you're including (or anything it includes) and goodbye 173 bytes of SRAM.

matthijskooijman, 2019-09-20:
I think this is something we cannot really fix, since it is just a limitation of the C/C++ language. In particular, it is pretty much impossible for the compiler to tell whether an ISR is actually enabled or not. We already support leaving out the Serial ISR (and related code) if you do not reference the Serial object at all, which I think is the best we can do.

If anyone has any ideas on how to fix this, I'm happy to hear them. Until then, I'm closing this issue.

closed 2019-09-20 by matthijskooijman

MCUdude, 2022-06-15:
@matthijskooijman sorry for bringing this one up after so long, but I've been trying to figure out the mechanism for optimizing out unused Serial ISRs. Where in the code or build system is this happening? I'm asking because I'd like to do the same in a library; leave out the ISR if it's not being used.

Thanks!

MCUdude, 2022-06-15:
For reference, the library I'm trying to modify is the [Logic library](https://github.com/MCUdude/MegaCoreX/tree/master/megaavr/libraries/Logic) A wapper for the CCL hardware for the megaAVR-0 series. The problem is that `ISR(CCL_CCL_vect)` is present in Logic.cpp, and is not optimized out even if the ISR is not in use. I've tried to move the ISR into its own .cpp file, but I still can't get the compiler to optimize it away

matthijskooijman, 2022-06-16:
@MCUdude, for some background on how this was implemented for HardwareSerial, see https://github.com/arduino/Arduino/pull/1711#issuecomment-29745866 and for a similar thing for the timer ISR, see https://github.com/arduino/ArduinoCore-avr/pull/320

One important bit here, is that this trick only works when linking through .a files. This is the default for the core, but libraries are linked through .o files by default (meaning each .o file of the library is specified explicitly on the linker commandline, and thus is included in the link unconditionally, and then the `used` attribute on the ISR prevents it from being optimized away again). You can change this using a value in `library.properties` (it's called `dot_a_linkage` I think, or something similar), which causes the library's .o files to be collected in a .a file, and that causes the linker to only include .o files that are actually needed to satisfy a missing symbol.

Hope this helps. If you have further questions, maybe good to open an issue in your library for further discussion? If you tag me, I'll find the issue from the notification :-)

MCUdude, 2022-06-18:
Thank you for your reply @matthijskooijman!
I ended up moving the ISR into its own cpp file, and with `dot_a_linkage=true`, it's being optimized out!