arduino/ArduinoCore-avr #269: Add -fshort-enums

NicoHood, 2019-09-16:
It took me an extremely long time to debug a c code that used an uint8_t enum (that requires the 8 bit and not 16 bit size). The reason is that `-fshort-enums` is not used for arduino. We should add this to the c and cpp flags, as you cannot define the enum size for c as in c++. The cpp still needs this flag, so if you include the header from a c++ file those are also compiled correct.

FYI that is what i use to check now in my source:

``` c
// Check if opcode enums are 8bit, otherwise the code will compile wrong
#ifndef __cplusplus
_Static_assert ((sizeof(flash_opcode_t) == sizeof(uint8_t)),
"Please use -fshort-enums to compile this source");
#endif
```

lxrobotics, 2019-09-17:
@NicoHood This is a good input. Could you create a small PR appending `-fshort-enums` within `platform.txt` ?

NicoHood, 2019-09-23:
@lxrobotics Sorry, aint got time for that.