arduino/ArduinoCore-avr #272: Improve default yield()

mikaelpatel, 2019-09-16:
Should the default yield() function implement a low power mode?

Currently yield() is an empty function and delay() will busy wait for the delay period. To help designers the default function could use the power management functions available and instead use a sleep mode and wait for the next interrupt. 

A typical implementation may be found in Cosa:
default_yield() https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/main.cpp#L154
Power::sleep() https://github.com/mikaelpatel/Cosa/blob/master/cores/cosa/Cosa/Power.cpp#L30

The Arduino core: 
yield() https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/hooks.c#L31
delay() https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/wiring.c#L106

Cheers!

lxrobotics, 2019-09-17:
Hi @mikaelpatel - you are completely right :smiley:. `yield()` is currently defined as a `weak` function which allows it to be overriden by a user (or a future Arduino multi-tasking implementation). `delay` is currently a busy-wait indeed. Since you seem to have quite a neat implementation in your `Cosa` core, why don't you stop by and create a pull-request? I'm looking forward to continoue this discussion within your PR and will therefore close this issue.

closed 2019-09-17 by lxrobotics

mikaelpatel, 2019-09-17:
The minimum change to the Arduino core would be to also make delay() a weak function. I think this was my intent from the beginning. The reference to Cosa was more a reference of how this could be used. Thanks!