arduino/ArduinoCore-avr #273: In Arduino micro/leonardo Pin HWB is not exposed as digital pin

elsp1991, 2019-09-16:
In the [Atmega32u4 Pin Mapping](https://www.arduino.cc/en/Hacking/PinMapping32u4) PE2 is documented as HWB so the expected behavior would be the user to be able to use it as a digital pin. 
In the official boards (leonardo & micro) is tied to ground but is totally usable if you are using it with a atmega32u4 board.
The pull down on this pin is not required as the default fuses do not have the HWBE bit set, meaning that is completely ignored during bootup.
So make sense to Included in the pins_arduino.h a mapping for as arduino digital pin 31 and Alias name HWB.

MarkJeronimus, 2019-10-16:
FYI

"HWB allows to execute the bootloader section after reset when tied to ground during external reset pulse. The HWB mode of this pin is active only when the HWBE fuse is enable. During normal operation (excluded Reset), this pin acts as a general purpose I/O." -- The 32u4 datasheet

ra1nb0w, 2023-02-09:
I made a patch that add support for HWB pin; it works on leonardo and micro.

```
--- variants/leonardo/pins_arduino.h.old	2023-02-09 11:22:14.996472892 +0100
+++ variants/leonardo/pins_arduino.h	2023-02-09 11:25:36.415048394 +0100
@@ -88,7 +88,7 @@
 #undef OCR2_6
 #undef OCR2_7

-#define NUM_DIGITAL_PINS  31
+#define NUM_DIGITAL_PINS  32
 #define NUM_ANALOG_INPUTS 12

 #define TX_RX_LED_INIT	DDRD |= (1<<5), DDRB |= (1<<0)
@@ -107,6 +107,8 @@
 #define LED_BUILTIN_RX 17
 #define LED_BUILTIN_TX 30

+#define HWB 31
+
 // Map SPI port to 'new' pins D14..D17
 #define PIN_SPI_SS    (17)
 #define PIN_SPI_MOSI  (16)
@@ -270,6 +272,8 @@
 	PB, // D28 / D10 - A10 - PB6
 	PD, // D29 / D12 - A11 - PD6
 	PD, // D30 / TX Led - PD5
+
+        PE, // D31 / HWB - PE2
 };

 const uint8_t PROGMEM digital_pin_to_bit_mask_PGM[] = {
@@ -308,6 +312,8 @@
 	_BV(6), // D28 / D10 - A10 - PB6
 	_BV(6), // D29 / D12 - A11 - PD6
 	_BV(5), // D30 / TX Led - PD5
+
+        _BV(2), // D31 / HWB - PE2
 };

 const uint8_t PROGMEM digital_pin_to_timer_PGM[] = {
@@ -346,6 +352,8 @@
 	NOT_ON_TIMER,
 	NOT_ON_TIMER,
 	NOT_ON_TIMER,
+
+        NOT_ON_TIMER,
 };

 const uint8_t PROGMEM analog_pin_to_channel_PGM[] = {
```