arduino/ArduinoCore-avr #277: eu_dst.h wrong calculation od Daylight Saving Time

vprheli, 2019-09-16:
Calculation use wrong conditions. Please correct
        if              (month == MARCH) {
            if (mday < d)
                return 0;
            if (mday > d)
                return ONE_HOUR;
            if (hour < 2)
                return 0;
            return ONE_HOUR;
        }
        if              (mday < d)
                            return ONE_HOUR;
        if              (mday > d)
                            return 0;
        if              (hour < 2)
                            return ONE_HOUR;
                        return 0; 

And check also usa_dst.h if there isn't similar error
Vladimir

facchinm, 2019-09-17:
@Robotto the only way is fixing it upstream in avr-libc (if it still applies). Could anyone provide a minimal and reproducible test to check if toolchain 7.3.0 is still affected?

Robotto, 2019-09-20:
Where is avr-libc being maintained?
I'm seeing it hosted in several places...

facchinm, 2019-09-20:
https://www.nongnu.org/avr-libc/ should be the project's home

Robotto, 2019-10-11:
It is a known bug, which i've added my input to here:
https://savannah.nongnu.org/bugs/?44327

There doesn't seem to be any way to do a pull request on the somewhat
archaic looking nongnu bugtracking system... :/
~Mark

Robotto, 2021-10-31:
This became relevant again at around 2 in the morning today ;)