arduino/ArduinoCore-avr #278: PROGMEM attribute ignored for implicit pointer initialization

alexbladex, 2019-09-16:
Windows Arduino IDE 1.8.9, default settings.

**Description:**
bellow both statements are identical. In both cases we get array of pointers, where all elements will be NULL pointers. And both variables occupies identical size 20 byte in SRAM
`const char *const list[10];`
`const char *const list[10] = {NULL};`

BUT

**Actual behavior:**
if we will try put it in Flash then for first one, PROGMEM attribute is ignored and it will be placed only in SRAM instead of PROGMEM
`const PROGMEM char *const list[10];  // will be placed in SRAM. progmem ignored`
`const PROGMEM char *const list[10] = {NULL};  // will be placed in PROGMEM`

**Expected behavior:**
`const PROGMEM char *const list[10];` should be placed in Flash

matthijskooijman, 2019-09-20:
I wonder if this is a gcc problem. @alexbladex, could you test this with the AVR core 1.8.1 again? It has a newer gcc version, so maybe this was fixed in there?

alexbladex, 2019-09-24:
with 1.8.0-1.8.2 : I get collect2.exe error ld returned 1 when compile. 
may be you mean 1.8.10 ?

matthijskooijman, 2019-09-24:
I mean AVR core 1.8.1 (which you can install through the boards manager inside the IDE), not the Arduino IDE 1.8.1. Does that clarify?

alexbladex, 2019-10-11:
> I mean AVR core 1.8.1 (which you can install through the boards manager inside the IDE), not the Arduino IDE 1.8.1. Does that clarify?

I have updated arduino AVR boards to 1.8.1 as you said

issue reproduced and in 1.8.1
progmem ignored
const PROGMEM char *const list[10]; // will be placed in SRAM. 
for more details pls refer to 1st post

oqibidipo, 2020-01-19:
This has nothing to do with PROGMEM. The only bug here is including `-fpermissive` in the AVR core options.

An uninitialized const is not valid in C++.

aa.cc:
```
#include <avr/pgmspace.h>

const char moo[10];
const int foo;
const long bar = 123456789;
```

```
> avr-gcc -mmcu=atmega328p -x c++ -fsyntax-only aa.cc
aa.cc:1:12: error: uninitialized const 'moo' [-fpermissive]
 const char moo[10];
            ^
aa.cc:2:11: error: uninitialized const 'foo' [-fpermissive]
 const int foo;
           ^
```
Adding `-fpermissive` turns these to warnings.
However, these are not errors in C:
```
> avr-gcc -mmcu=atmega328p -x c -fsyntax-only -Wall -Wextra aa.cc
(no output)
```
`avr-gcc` versions tested: 4.8.1, 5.4.0 7.3.0, 9.1.0.
The only difference is how the messages are decorated.

This is just another example why having `-fpermissive` enabled by default is plain EVIL.