arduino/ArduinoCore-avr #282: Adding SPI programming on a Raspberry Pi (from the IDE)

lewispg228, 2019-09-17:
I've been working on a [stand-alone AVR programming solution](https://learn.sparkfun.com/tutorials/raspberry-pi-stand-alone-programmer) for SparkFun over the past couple years. Currently, we use Python (watching a GPIO) to engage a bash file that calls avrdude.  We have found that the SPI hardware built into the raspberry pi is very reliable and it would be awesome to be able to also develop within the arduino IDE (on a raspi), and upload via SPI.

Would this be possible? Is it a crazy request?

lxrobotics, 2019-09-17:
@per1234 can you please prepare such a PR?

adrienlaveau, 2021-05-30:
Hello everyone,

I am looking to flash an ATmega328p with my raspberry Pi using the Arduino IDE. Is the solution available ? Or should we export the binary and flash through the terminal ?

Thanks