arduino/ArduinoCore-avr #283: ++ operator used for array index ( array[index++] )

ionicat, 2019-09-17:
After executing the following code:

	int i=0;
	char s1[20];
	char s2[20];
	strcpy(s1, "test1");
	strcpy(s2, "xxxxxx");
	while(i<4)
		s2[i++] = s1[i];
	s2[i++] = '2';
	s2[i] = 0;
on Visual Studio (Windows) 
s1 = "test1"
s2 = "test2"
as expected
however, on Arduino
s1 = "test1"
s2 = "est12"
Arduino acts like 

	while(i<4)
		s2[i++] = s1[i];
is equivalent with

	while(i<4){
		s2[i] = s1[i+1];
		i++;
	}
instead of

	while(i<4){
		s2[i] = s1[i];
		i++;
	}

on Arduino, add
```
Serial.println(s1); 
Serial.println(s2); 
```
to see string values in Serial Monitor window

This is just a simplified code to show the issue discovered in a more complex situation, I know that something like

	int i=-1;
	while(i++<4)
		s2[i] = s1[i];
or a "for" instruction would have been fine

I have changed to

	while(i<4)
		s2[i] = s1[i++];
and got the expected result, good to know

cmaglie, 2019-09-17:
Barely looking at your code my expected output is `est12`... but to be sure I tried the following on my linux machine:

```
:~$ cat test.c 
#include <stdio.h>
#include <string.h>

void main() {
	int i=0;
	char s1[20];
	char s2[20];
	strcpy(s1, "test1");
	strcpy(s2, "xxxxxx");
	while(i<4)
		s2[i++] = s1[i];
	s2[i++] = '2';
	s2[i] = 0;
	printf("%s\n%s\n", s1, s2);
}

:~$ gcc test.c -o test
:~$ ./test 
test1
est12
```

In fact `s2[i++] = s1[i]` is actually equivalent to:

```
t = i;
i++;
s2[t] = s1[i];
```

So it looks like a bug in your code than in Arduino. There is a reason why using prefix-increment like `i++` is discouraged inside other expression: because the result may be non immediately obvious.

For the next time, please use the forums to get help in programming tasks
-> https://forum.arduino.cc

Post issues here only when you are 100% that there is a problem with the IDE / Core.

closed 2019-09-17 by cmaglie

ionicat, 2019-09-18:
I have tested the same code on a Windows machine and that was my reference. What I see now is the same code compiled with different compilers is giving different results and I guess depending on from where are you coming the other one is the one having a bug.

cmaglie, 2019-09-18:
The meaning of

  `s2[i++] = s1[i]`

is

  `s2[i] = s1[i+1]; i=i+1`

there is no difference between compilers.

ionicat, 2019-09-18:
Ok then my Microsoft C++ compiler is wrong for not respecting the meaning of
s2[i++] = s1[i]
and compiling it as
s2[i] = s1[i]; i=i+1

cmaglie, 2019-09-18:
> Ok then my Microsoft C++ compiler is wrong for not respecting the meaning of

Yes.

But I'll bet that it's not your compiler that is wrong, but it's you that have messed up something during your tests.

I'll invite you to check again more carefully and see the results again.

See also https://stackoverflow.com/a/34251125 for an explanation on why it's like that.

Said that, I'll not reply anymore on this thread, if you want to follow my hint or not it's up to you, I'll not gain anything from this discussion :-)

ionicat, 2019-09-18:
I can not confirm your bet (see attached image) but I agree to leave it to that, we have spent to much time on this :-)
[image]