arduino/ArduinoCore-avr #285: Was there a mistaken release? or Where is development? and What's v1.8.1 of this core?

greyltc, 2019-09-18:
[image]

My official IDE just asked me to update the "Arduino AVR Boards by Arduino" library from 1.6.23 to 1.8.1. I don't see anything about 1.8.1 here so I'm really confused as to where the development for the 1.8.1 release that just hit my IDE came from or even what changes are in it.

Was the update my IDE got just some version number/release mistake or something?

facchinm, 2019-09-18:
That's normal, the dropdown doesn't list the already installed versions :slightly_smiling_face: 
It you downloaded IDE 1.8.10 (which contains avr core 1.8.1) it's already installed and the "update" button will just remove the old local version (if present).

facchinm, 2019-09-18:
Ah, but you are right, there's no commit about 1.8.1 (and no tag) since it's a repackaged 1.6.23 but using the new toolchain; I'm adding the tag right now

closed 2019-09-18 by facchinm

greyltc, 2019-09-18:
Yeah. I'm not commenting on the dropdown not having the version number. I was commenting on 1.8.1 hitting my IDE without anything in this repo being 1.8.1

facchinm, 2019-09-18:
Yup, I just pushed the tag, should be fine now