arduino/ArduinoCore-avr #290: Loosing last byte with readBytesUntil

danielkucera, 2019-09-22:
I have following code:
```
            client_len = serverClients[i].readBytesUntil(_EOF, client_buf, MAX_FRAME_LENGTH);
            client_buf[client_len++] = serverClients[i].read();
            if (client_buf[client_len-1] != _EOF){
              //something strange happened
              serverClients[i].write("uhm");
              continue;
            }
```
How should I distinguish if _EOF character or MAX_FRAME_LENGTH or timeout has been reached?
The implementation reads the terminating character but then it is lost (not written to output buffer):
https://github.com/arduino/ArduinoCore-avr/blob/317513b9c989de24d438de7ac55d701b738881e2/cores/arduino/Stream.cpp#L224

See also: https://github.com/esp8266/Arduino/issues/6546

UtkarshBhatthere, 2020-03-23:
Hey,
The timeout is implemented in the Stream::timedRead() on per character basis, if timeout occurs then the index value (in the implementation) doesn't get incremented hence the loop will last until MAX_FRAME_LENGTH characters are read, also since the terminator character is read by the Stream::ReadBytesUntil() api it's not present in the Stream anymore hence 
client_buf[client_len++] = serverClients[i].read();
will not read _EOF since it's already been read by the previous call to Stream::ReadBytesUntil().

The best shot is to compare the client_len and MAX_FRAME_LENGTH, if they're equal then maximum length was read otherwise _EOF was encountered.

danielkucera, 2020-03-23:
I my opinion, the best solution would be to write the terminating character to `buffer` like this:
```
if (c < 0) break;
*buffer++ = (char)c;
if (c == terminator) break;
```

UtkarshBhatthere, 2020-03-23:
I think the implementation intentionally doesn't read the terminator character, i.e. it's by design. As long as we have methods to know if terminator was encountered changing the implementation isn't necessary. It would also break codebases written with an expectation that it will not include the terminator character.

danielkucera, 2020-03-23:
What methods do we have to know if terminator was encountered?