arduino/ArduinoCore-avr #294: Unable to compile lmic lib with Arduino AVR Boards 1.8.1

vlastahajek, 2019-10-03:
After I've upgraded Arduino IDE to 1.8.10, which in turn brought Arduino AVR Boards 1.8.1, I'm unable to compile sketch, which uses [lmic libary](https://github.com/hallard/arduino-lmic):
```
C:\Users\vlast\AppData\Local\Temp\ccyy7JQt.ltrans0.ltrans.o: In function `radio_irq_handler':

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/radio.c:801: undefined reference to `table_get_u2'

C:\Users\vlast\AppData\Local\Temp\ccyy7JQt.ltrans0.ltrans.o: In function `schedRx12':

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:1336: undefined reference to `table_get_ostime'

C:\Users\vlast\AppData\Local\Temp\ccyy7JQt.ltrans0.ltrans.o: In function `calcBcnRxWindowFromMillis':

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:419: undefined reference to `table_get_ostime'

C:\Users\vlast\AppData\Local\Temp\ccyy7JQt.ltrans0.ltrans.o: In function `calcRxWindow':

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:405: undefined reference to `table_get_ostime'

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:407: undefined reference to `table_get_ostime'

C:\Users\vlast\AppData\Local\Temp\ccyy7JQt.ltrans0.ltrans.o: In function `initDefaultChannels':

C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:553: undefined reference to `table_get_u4'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "lmic.h"
 Used: C:\Users\vlast\OneDrive\Dokumenty\Arduino\libraries\arduino-lmic
Multiple libraries were found for "SPI.h"
 Used: C:\software\arduino.1.8.10\hardware\arduino\avr\libraries\SPI
exit status 1
Error compiling for board Arduino Pro or Pro Mini.
```
Works well again after I had downgraded back to 1.6.23

matthijskooijman, 2019-10-03:
This is probably not an Arduino problem, but a change in the compiler (gcc was upgraded internally) that exposes a bug or other problem in the library. I had already noticed this problem a few days ago, but haven't had a chance to look at it.

Could you perhaps report this at https://github.com/matthijskooijman/arduino-lmic ? Then I'll see if I can fix this in the near future. In the meanwhile, I'm closing this issue, since I do not think this is something Arduino can fix (and if I investigate and conclude otherwise, we can always reopen).

closed 2019-10-03 by matthijskooijman

vlastahajek, 2019-10-03:
Ok, thanks @matthijskooijman  for the quick feedback. I will report it. 
Just for explanation, I referenced Hallard's fork cause I'm using it for it's support for pulled interrupts on LoRa Radio Node v1.0