arduino/ArduinoCore-avr #295: how to change the usb name of Leonardo\ProMicro

KarlSteeg, 2019-10-07:
hi there.

please can someone advise how to change the USB/Friendly name on the leonardo ProMicro u32's?

i have a few arduinos working, and would like to name them after function to avoid incorrect sketch loading (the all show as arduino nano and leonardos)

KarlSteeg, 2019-10-18:
Hey Jeffrey

Awesome thank you, 
i was suspecting that line but unsure.

im i understanding correctly that, that change can be performed with sketch upload?
and is ther specific names i must use?
Can i call it generic names? eg Sim Pedals or Arduino G27 Pedals.
or does it requires set names?   eg Joystick as pposed to joystick.

many thanks
karl

PaulStoffregen, 2019-10-18:
If you use Microsoft Windows, the name you see in the Device Manager and Windows notifications depends on the INF file which Windows associates with the device or its interfaces.  Changing the USB descriptors only changes the name users actually see on Linux and Macintosh.

The INF is just a text file.  Creating one is fairly easy if you just copy an example and edit the names.  But all modern versions of Windows require a corresponding CAT file which matches the INF and have a valid digital signature.  You create this signature using a couple command line programs which are part of the free WDK, but to do so you must have a valid code signing certificate which is not free.  The best deal I've found is [K Software](https://www.ksoftware.net/code-signing-certificates/), at about $80.  Even with Windows 10, you only need the cheaper OV type certificate to sign INFs.  You only need the expensive EV cert if you write actual driver code in a SYS file.

JeffreyO, 2019-10-19:
There is another redundant one in bootloaders/caterina/Descriptors.c

closed 2022-01-03 by per1234