arduino/ArduinoCore-avr #298: Issue with having multiple analogWrite's in 1.8.1

SonnyX, 2019-11-20:
The following code does not work in 1.8.1
```c
long PosX = 0;
long TargetX = 0;

void setup() {
  Serial.begin(115200);
  analogWrite(9, 0);
}
 
void loop() {
  if(PosX != TargetX) {
    analogWrite(9,200);
  } else {
    analogWrite(9,200);
  }
}
```
The following does:
```c
long PosX = 0;
long TargetX = 0;

void setup() {
  Serial.begin(115200);
  analogWrite(9, 0);
}
 
void loop() {
  if(PosX != TargetX) {
    //analogWrite(9,200);
  } else {
    analogWrite(9,200);
  }
}
```
This issue does not exist in 1.6.23

facchinm, 2019-11-20:
Hi @SonnyX ,
what do you mean with 1.8.1 ? Which board are you targeting? And which behaviour are you observing? Attaching the verbose output for both compilations could help :slightly_smiling_face:

SonnyX, 2019-11-20:
I am talking about `Arduino AVR Boards` by `Arduino` in the Boards Manager.
`1.8.1` should kind of make sense, as the last release in this repository was also named `1.8.1`.
The code was ran on an `atMega2560`.

facchinm, 2019-11-20:
Ok, thanks. IDE versions and AVR core are getting similar these days :slightly_smiling_face: 
I tried your code (the one that should fail) with a UNO and it works with core 1.8.1 .
As soon as I grab a Mega2560 I'll be testing it too.