arduino/ArduinoCore-avr #306: Feature Request: Instructions for offline installation

marc1uk, 2019-12-30:
We just purchased an arduino every for use in a high voltage control system. The system is on a secure network without internet access. While installing the IDE was straightforward, installing the avr core libraries seems to be online-only. Forum suggestions primary seem to require obtaining a second windows machine with internet access from which to do a online installation, then copy over the files. Hardly practical.
Currently I'm struggling with some rather terse ad unclear instructions from a forum user, but it's a pretty poor user experience so far. 
These are essential installation steps; they should not depend on an internet connection.

facchinm, 2020-01-09:
Hi @marc1uk ,
the AVR core is included in the package we distribute. Can you paste a screenshot of your `Tools/Boards` menu?

deepaklorkhatri007, 2020-02-29:
> the AVR core is included in the package we distribute.

Hi @facchinm, We don't get the support for Arduino Every board when we do a fresh install of Arduino. 

> The system is on a secure network without internet access

Hi @marc1uk, I suppose you can use a USB stick for bringing files to your system which is without internet access, If that's the scenario I suggest you use the [poratble](https://www.arduino.cc/en/Guide/PortableIDE) version of the Arduino IDE, just install the Arduino Every board and other libraries you want :)

facchinm, 2020-03-02:
Nano Every is not part of AVR core but megaAVR indeed, so it needs to be downloaded separately. Indeed the portable version, created on a computer with internet capabilities, will solve the issue.

closed 2020-03-02 by facchinm

marc1uk, 2020-03-07:
I really don't think this "solves" the issue. I included in the question this suggestion and remarked that it is not always practical. Responding to 

> how do i install nano Every drivers without an internet connected windows machine

with
> download them with your internet connected windows machine

is not an answer. As ubiquitous as some may think Windows is, many industries use Linux/macOS. The suggestion amounts to "buy a windows laptop and take it off-site", which - as I said - is hardly practical.

deepaklorkhatri007, 2020-03-07:
@marc1uk It's not possible to ship the Arduino IDE with support for all the available boards out there. that will make it a huge package, please try the [portable](https://www.arduino.cc/en/Guide/PortableIDE) version of the Arduino IDE.

> I really don't think this "solves" the issue. I included in the question this suggestion and remarked that it is not always practical. Responding to

I don't think it's an issue of Arduino but a personal one. You have created this issue for getting help on solving your problem and we have given a solution anyway, use it wisely. 

> As ubiquitous as some may think Windows is, many industries use Linux/macOS. 

Do you really want to get it working? I shared the link for a portable version of the Arduino, have you opened the link for once? It clearly shows the option for making portable version of IDE for Linux.

marc1uk, 2020-03-10:
I agree that shipping the IDE with support for all available boards would make it bloated and is undesirable. But perhaps they could be provided as separate downloads? 
I did look at the page for the generation of a portable IDE, and it is indeed very straightforward, full credit for a job well done.

Sadly it does not work around my limitation. My remark about Windows in particular was too narrow. Rather, extra board support can only be obtained with an internet connected computer of the same OS as the destination platform. I presume building a portable version of the IDE on our internet connected machines (linux) would not work on the client machine (windows).

Perhaps you are right, though. This is a niche scenario. A "personal issue". I will do some self-reflection. Perhaps I will have an epiphany.

In the meantime my epiphany was to use another device. You're also right that my follow-ups here are largely academic. This issue was opened in December, and we have since moved to using a National Instruments NI-USB. Device drivers included. :+1: