arduino/ArduinoCore-avr #314: gcc 7.3.0 "redeclared as different kind of symbol" bug

talbotsk, 2020-02-26:
Compiler confuses between completely different types of symbols. Example:

typedef enum ECommands
{
  Play
};

void Play()
{
}

ECommands command = ECommands::Play;

void setup()
{
}

void loop()
{
  switch (command)
  {
    case ECommands::Play:
      Play();
      break;
  }
}

Compiler reports "'Play' redeclared as different kind of symbol" error even it should not be an error.

talbotsk, 2020-02-27:
As I read in ANSI/ISO specification of C/C++ languages the identifier of enum member can be used without the scope identifier of the enum (the name of the enum). Therefore the compiler correctly reports the error during the compilation.

closed 2020-02-27 by talbotsk

matthijskooijman, 2020-02-27:
@talbotsk You might want to look at C++ "enum class", which does encapsulate the enum values in a new scope (but does have some other implications as well).