arduino/ArduinoCore-avr #323: Serial.read() corrupted with high baudrate

landret, 2020-03-26:
Hi,
I have noticed a problem which occurs with Baudrate above 115200 (for instance 140000, 230400 or  250000). From time to time, data received by Arduino may become corrupted with data sent by Arduino. Substring of sent data may appear in received data.

**Testing conditions:**
- Arduino is repeatedly sending data using Serial.print() : “TX:AAAA”
- Arduino is punctually receiving data from computer :  “BBBBBBBBBB” (no CR no LF)
- Received data is stored in a buffer that is then print in Serial monitor using Serial.print() : “RX: + (length of buffer) + “_buffer”

**Example of Serial monitor:**

**Good behavior:**
TX:AAAA
TX:AAAA
RX:2_BB
RX:8_BBBBBBBB
TX:AAAA
TX:AAAA
TX:AAAA
TX:AAAA
TX:AAAA
TX:AAAA
TX:AAAA
RX:10_BBBBBBBBBB
TX:AAAA
TX:AAAA

**Bad behavior (from time to time)**
TX:AAAA
TX:AAAA
RX:5_BBBBB
TX:AAAA
RX:5_B:ABB
TX:AAAA
TX:AAAA
TX:AAAA
TX:AAAA
TX:AAAA
TX:AAAA
RX:4_BBTX
RX:6_:AAABB
TX:AAAA
TX:AAAA

```C++
char rx_buffer[60];
byte index = 0;

void setup()
{
    Serial.begin(230400);
}

void loop()
{
    if (Serial.available())
    {
        // read input
        while (Serial.available() > 0)
        {
            rx_buffer[index] = Serial.read();
       
            if (index < 59)
                index++;
        }
        
        // clear buffer end
        for(byte i = index; i < 60; i++)
            rx_buffer[i] = '\0';

        // reset index
        index = 0;

        Serial.print("RX:");
        Serial.print(strlen(rx_buffer), DEC);
        Serial.print("_");
        Serial.println(rx_buffer);
    }
    else
        // write
        Serial.println("TX:AAAA");
}
```

This problem happens whatever the data length received or sent.
Also, this problem does NOT happen if the rx_buffer has fixed value and is not written to by Serial.read(). For instance, doing the following will not recreate the problem:

```C++
char rx_buffer[60] = "BBBBBBBBBB";
byte index = 0;

void setup()
{
    Serial.begin(230400);
}

void loop()
{
    if (Serial.available())
    {
        // read input
        while (Serial.available() > 0)
            Serial.read();

        Serial.print("RX:");
        Serial.print(strlen(rx_buffer), DEC);
        Serial.print("_");
        Serial.println(rx_buffer);
    }
    else
        // write
        Serial.println("TX:AAAA");
}
```

Then this is not an EMI problem but rather a software problem. I suspect that Serial.read() may return corrupted data.

matthijskooijman, 2020-03-27:
Looked through your code, could not find anything weird there.

What board and Arduino core are you using? And what are you running on the host to send the data exactly? These will help reproducing.

landret, 2020-03-27:
Tested with official Arduino Uno and Mega, same results.
I am using the Terminal Program RealTerm on the host computer to send and monitor data. I send the data ("BBBBBBBBBB" no CR no LF) manually by pressing button send. Phenomenon is more prone to occur when sending data quickly and repeateadly but it can also happens with isolated sending.
I also tested this with other programs, same results happens.

matthijskooijman, 2020-03-27:
> Then this is not an EMI problem but rather a software problem. I suspect that Serial.read() may return corrupted data.

I'm not sure this is entirely true: it might still be an EMI problem on *reception* rather than transmission. I don't think it is, but might be.

Something interesting to try would be to modify your code to count the number of B's in the rx buffer and print that as well, then you'll know if the problem occurs on reception or transmission.

Are you using the latest Arduino IDE and AVR core (see board manager)?

matthijskooijman, 2020-03-27:
I just tried this on an Uno, with AVR core 1.8.12, with minicom on the host and could reproduce this as well (just keep B pressed on the console). Had to enable logging to see what happened, and it seems minicom was too busy with printing the TX's to send a lot of B's, but the behaviour does show. Here's an excerpt (showing only `grep ^RX` lines):

```
RX:1_B
RX:1_A
RX:1_B
RX:1_B
RX:1_B
RX:1_B
RX:1_T
RX:1_B
RX:1_A
RX:1_B
RX:1_B
RX:1_B
RX:1_:
RX:1_B
RX:1_B
RX:1_B
RX:1_B
RX:1_A
RX:1_B
RX:1_B
RX:1_B
RX:1_B
RX:1_B
RX:1_B
RX:1_B
RX:1_B
RX:1_B
RX:1_X
```

matthijskooijman, 2020-03-27:
Ok, counting the number of B's in the sketch shows that the problem happens in reception, since the rx_buffer actually contains the wrong data.

I used this code:

```
    int bs = 0;
    for (byte i = 0; i < 60 && rx_buffer[i]; i++)
      if (rx_buffer[i] == 'B') ++bs;
    
    Serial.print("RX:");
    Serial.print(strlen(rx_buffer), DEC);
    Serial.print("_");
    Serial.print(bs, DEC);
    Serial.print("_");
    Serial.println(rx_buffer);
```

and got this result:

```
RX:1_1_B
RX:1_0_T
RX:1_1_B
RX:1_1_B
RX:1_1_B
RX:1_0_A
RX:1_1_B
RX:1_1_B
RX:1_1_B
RX:1_1_B
RX:1_0_A
RX:1_1_B
RX:1_1_B
RX:1_0_
```

However, it seems this is actually known bug already, in the 16u2 firmware. It was previously reported and analyzed here: https://github.com/arduino/ArduinoCore-avr/issues/296 so I'm going to close this bug as a duplicate, and maybe poke at that bug a bit more.

closed 2020-03-27 by matthijskooijman

landret, 2020-03-27:
Thanks Matthijs for your investigation