arduino/ArduinoCore-avr #330: Feature Request/Solution Proposed: Support all pins of Atmega 1280/2560

chepo92, 2020-04-14:
There are 10 pins in the Atmega 1280/2560 that are unmapped from digital pin to port, I know it would not work with official arduino mega boards, but will do with other custom boards and solve some issues (like a Mightyboard controller), https://github.com/MarlinFirmware/Marlin/issues/15607
I think this should work
https://github.com/chepo92/ArduinoCore-avr/blob/master/variants/mega/pins_arduino.h
Should I submit a PR? to which branch? 
or discuss it in the mailing list? 
Maybe create another board?

per1234, 2020-04-14:
If this is to be done, I suggest that we consider using the already well established pin mappings from MCUdude/MegaCore to avoid the proliferation of variants:
https://github.com/MCUdude/MegaCore/blob/master/avr/variants/100-pin-arduino-mega/pins_arduino.h#L220-L235

chepo92, 2020-04-14:
Nice, I didn't know the existence of that repo, there must be some other variants.
Maybe the modification/variant is not needed, as MCUdude/MegaCore seems a good "standard" I'll give it a try, but I dont know if it is constantly maintained and synced with the official core (and fully compatible with libraries) 
Maybe modifying original pins is not possible as there may be a code that uses NUM_DIGITAL_PINS or other variables in the pins file with the original Board.

closed 2020-12-10 by chepo92