arduino/ArduinoCore-avr #333: shiftIn loses the first bit with CD4021BE

dzervas, 2020-04-26:
The clocking though seems to be wrong. Instead of <clock high><read><clock low>, it should be <read><clock low><clock high>. At least that's what is working with my CD4021BE, otherwise I'm losing the first bit.

matthijskooijman, 2020-04-27:
Hm. I was planning to write that this just depends on the hardware you're using and that `shiftIn` was just documented and intended like this, and probably works fine with other hardware (which is probably true for hardware that needs a clock rising edge to output the first data bit, unlike the 4021, which outputs the first bit right after the latch).

However, looking at https://www.arduino.cc/en/tutorial/ShiftIn initially suggests that `shiftIn` should just work with a 4021 as well. Looking more closely shows that the tutorial code actually implements `shiftIn` in the sketch, possibly predating the inclusion of `shiftIn` in the core? Anyway, the tutorial version indeed pulls clock low first, then reads and only pulls clock high after each bit.

So maybe `shiftIn` is indeed not optimal like it is now, but changing it would also be painful (there might be sketches that rely on the current behaviour, and I suspect all third-party cores will have copied the faulty behaviour too).

As a workaround, I think you can maybe write the clock pin high before latching and before calling `shiftIn`. Looking at [how `shiftIn` is implemented](https://github.com/arduino/ArduinoCore-avr/blob/2f67c916f6ab6193c404eebe22efe901e0f9542d/cores/arduino/wiring_shift.c#L25-L38), it seems it starts by making clock `HIGH`, without pulling it `LOW` first, so making clock `HIGH` beforehand would skip the first rising edge.

If this indeed works, I think this would be good to explicitly document in the `shiftIn` reference page, maybe adding an example for the 4021 even. If you're interested, you could submit a pullrequest for that at https://github.com/arduino/reference-en

dzervas, 2020-05-24:
You're actually right, you shouldn't change the implementation. I'm closing this. Thank you!

closed 2020-05-24 by dzervas